xdspacelab/openvslam #154: use ndk in android complie error with g2o

jinkailong888, 2019-10-23:
CMakeFiles/openvslam.dir/src/main/jni/openvslam/src/openvslam/optimize/pose_optimizer.cc.o: In function `openvslam::optimize::pose_optimizer::optimize(openvslam::data::frame&) const':
/Users/jinkailong/Documents/VR/openvslamtestnew/app/.cxx/cmake/debug/arm64-v8a/../../../../src/main/jni/openvslam/src/openvslam/optimize/pose_optimizer.cc:30: undefined reference to `g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(std::__ndk1::unique_ptr<g2o::Solver, std::__ndk1::default_delete<g2o::Solver> >)'
CMakeFiles/openvslam.dir/src/main/jni/openvslam/src/openvslam/optimize/local_bundle_adjuster.cc.o: In function `openvslam::optimize::local_bundle_adjuster::optimize(openvslam::data::keyframe*, bool*) const':
/Users/jinkailong/Documents/VR/openvslamtestnew/app/.cxx/cmake/debug/arm64-v8a/../../../../src/main/jni/openvslam/src/openvslam/optimize/local_bundle_adjuster.cc:104: undefined reference to `g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(std::__ndk1::unique_ptr<g2o::Solver, std::__ndk1::default_delete<g2o::Solver> >)'
CMakeFiles/openvslam.dir/src/main/jni/openvslam/src/openvslam/optimize/local_bundle_adjuster.cc.o: In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solveBlocks(double**&, g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&)':

use ndk in android complie error

jinkailong888, 2019-11-05:
g2o Compilation problem

closed 2019-11-05 by jinkailong888

saqib1707, 2020-08-20:
@jinkailong888 I am also facing this issue for some time. I have tried various ideas after searching on internet but unable to resolve this. As your previous comment says "g2o compilation problem", can you please provide more details regarding the compilation issue and how exactly you solved it ?