xdspacelab/openvslam #155: Build without Pangolin and without SocketViewer

ojjsaw, 2019-10-24:
Errors when doing make without above two (i.e. OFF for both) with modified cmakelist.txt in examples directory.
[image]

shinsumicco, 2019-11-05:
@ojjsaw 
It seems some library is not linked to executable you want to build.
Please check instructions about it.

chengwei-peng, 2019-11-29:
CMAKE_OPTIONS += -DBUILD_WITH_MARCH_NATIVE=OFF \
				-DUSE_PANGOLIN_VIEWER=OFF \
				-DUSE_SOCKET_PUBLISHER=OFF \
				-DUSE_STACK_TRACE_LOGGER=ON \
				-DBOW_FRAMEWORK=DBoW2 \
				-DUSE_G2O_DEBUG=OFF \
				-DBUILD_TESTS=OFF

shinsumicco, 2019-11-29:
@chengwei-peng Solved?

Shashika007, 2019-11-29:
@shinsumicco @chengwei-peng @ojjsaw  it works!  
cmake   CMAKE_OPTIONS +=   -DBUILD_WITH_MARCH_NATIVE=ON     -DUSE_PANGOLIN_VIEWER=OFF     -DUSE_SOCKET_PUBLISHER=OFF     -DUSE_STACK_TRACE_LOGGER=ON     -DBOW_FRAMEWORK=FBoW ../

chengwei-peng, 2020-01-04:
ok,thanks!