xdspacelab/openvslam #156: The camera coordinate of keypoints.

JinglingZhu, 2019-10-24:
Thanks for this great project. It is so friendly for development.

I can get all the camera pose easily. Currently, I can get 2D coordinates (seems like pixel coordinates) via `curr_frm_.keypts_.at(idx).pt`.
But is there any possible to output the camera coordinates of keypoints?

The background: 
I'd like to use SLAM to localization indoor in Architecture & Construction area. It is required to match with coordinates in BIM (Building Information Model, usually use geographic coordinates). 
My preliminary solution is that I can use some existing benchmarks (Markers with geographic coordinates) to match with keypoints of openvslam. It means if I can find the marker center as a keypoint in SLAM. I can match the marker's geographic coordinate with keypoint's camera coordinate. So I can get  geographic coordinates of all camera poses. Then I can capture the camera view in BIM with same camera location.

Thanks!

JinglingZhu, 2019-10-28:
I want to get some coordinates pairs (2D pixel coordinate of keypoints, world coordinate of landmark). I assume that I can get the two coordinates from one keypoint/landmark by the same index. For example, I assume `curr_frm_.keypts_.at(0).pt` and `curr_frm_.landmarks_.at(0)->get_pos_in_world()` reprensent one point.
I did some test, the size of keypoints vector and landmarks vector in the curr_frm_ are equal. I can get all the coordinates of every keypoints (2D) via `curr_frm_.keypts_.at(idx).pt,` but I found not every landmark can output coordinates value successfully via `curr_frm_.landmarks_.at(idx)->get_pos_in_world()`. The object `curr_frm_.landmarks_.at(idx)` is often null. Would you please give some advises because I really need this work fine.

Thanks!

shinsumicco, 2019-11-05:
@JinglingZhu 
Thank you for your interest in this project.
If a valid landmark for a 2D keypoint doesn't exist, the pointer of the keypoint will be `nullptr`.
So, you need to null-check before accessing to a landmark coords via `get_pos_in_world()`.

rahuldeo2047, 2020-04-06:
For better issue referential tracking : https://github.com/xdspacelab/openvslam/issues/157

mirellameelo, 2020-08-26:
@JinglingZhu @shinsumicco Hi!

Tks for sharing your answers! It helped me a lot. 
I was using the function "get_valid_landmarks()" because it disregards the _nullptr_. But, I couldn't find a way to associate it with the _keypoints_. So, just to make sure, using "get_landmarks()" instead of "get_valid_landmarks()", I can conclude that landmark.at(idx) are "made" from keypoints.at(idx), right? 

And just one more question, you said that some times _curr_frm_.keypts_.at(0).pt and curr_frm_.keypts_right_.at(0).pt are total different point_. Did you find any solution or explanation for that? 

Tks in advance. :)

ymd-stella, 2020-08-27:
> I was using the function "get_valid_landmarks()" because it disregards the nullptr. But, I couldn't find a way to associate it with the keypoints. So, just to make sure, using "get_landmarks()" [...]

Since it is calculated as follows, I think `landmark.at(idx)` are made from `keypoints.at(idx)`.

1. frame::keypts_ and frame::descriptors_ are computed by [orb_extractor::extract](https://github.com/xdspacelab/openvslam/blob/5919b408aff485b89a85710b9e2be29ccb7803ee/src/openvslam/feature/orb_extractor.cc#L68) in initialization.

2. frame::bow_feat_vec_ is computed from frame::descriptors_ by [TemplatedVocabulary<TDescriptor,F>::transform](https://github.com/dorian3d/DBoW2/blob/3924753db6145f12618e7de09b7e6b258db93c6e/include/DBoW2/TemplatedVocabulary.h#L1125) in 
[frame::compute_bow](https://github.com/xdspacelab/openvslam/blob/5919b408aff485b89a85710b9e2be29ccb7803ee/src/openvslam/data/frame.cc#L165).
frame::compute_bow is called by [frame_tracker::bow_match_based_track](https://github.com/xdspacelab/openvslam/blob/6493600a8f89a132e665667002a05669783d62e6/src/openvslam/module/frame_tracker.cc#L61) or [relocalizer::relocalize](https://github.com/xdspacelab/openvslam/blob/5919b408aff485b89a85710b9e2be29ccb7803ee/src/openvslam/module/relocalizer.cc#L28).

3. matched_lms_in_frm is computed from bow_feat_vec_ in [bow_tree::match_frame_and_keyframe](https://github.com/xdspacelab/openvslam/blob/5919b408aff485b89a85710b9e2be29ccb7803ee/src/openvslam/match/bow_tree.cc#L16), and set to frame::landmarks_ in [frame_tracker::bow_match_based_track](https://github.com/xdspacelab/openvslam/blob/5919b408aff485b89a85710b9e2be29ccb7803ee/src/openvslam/module/frame_tracker.cc#L73).

> And just one more question, you said that some times curr_frm_.keypts_.at(0).pt and curr_frm_.keypts_right_.at(0).pt are total different point. Did you find any solution or explanation for that?

Please refer comment https://github.com/xdspacelab/openvslam/blob/5919b408aff485b89a85710b9e2be29ccb7803ee/src/openvslam/data/frame.h#L186-L189.

mirellameelo, 2020-08-27:
@ymd-stella thank you so much!