xdspacelab/openvslam #157: ROS | How to get 2D ot 3D map out from openvslam?

rahuldeo2047, 2019-10-25:
I checked some openvslam code and the ROS package. I could not find any or similar reference for ROS topics for following

`observation_sources: laser_scan_sensor point_cloud_sensor`

Any help in that direction? Let me know if any info required.

I want to use a mono camera and map&localize my mobile robot using openvslam. This research looks very promising. 

(Camera calibration will do later. https://github.com/xdspacelab/openvslam/issues/104)

rahuldeo2047, 2019-10-26:
May be invalid issue. Will check back.

rahuldeo2047, 2019-11-13:
Is this issue relevant?

klintan, 2020-01-15:
I created a very basic example for odometry in ROS2 https://github.com/xdspacelab/openvslam/pull/225
Feel free to check it out and add your feedback.

rahuldeo2047, 2020-01-27:
Thank you @klintan for https://github.com/xdspacelab/openvslam/pull/225
I'll check it ASAP.

rahuldeo2047, 2020-01-27:
Perhaps, I will have to change it to ROS1 first.

rahuldeo2047, 2020-02-18:
**This is a status update and yet to verify the output.**

@klintan : I gave one Sunday to it.
Converted it to ROS1. Getting Odom. However, I am still to verify the output w.r.t. the video and the real world.

my learning for the API part

```
openvslam::system SLAM(cfg, vocab_file_path);

std::shared_ptr<openvslam::publish::map_publisher> map_publisher_ = SLAM.get_map_publisher();

auto cam_pose_ = map_publisher_->get_current_cam_pose();
```
However, 
`SLAM.feed_monocular_frame(cv_bridge::toCvShare(msg, "bgr8")->image, timestamp, mask);`
returns the `auto cam_pose_` too.

Is there any specific reason for calling `SLAM.get_map_publisher();` and `map_publisher_->get_current_cam_pose();`?

klintan, 2020-02-19:
> **This is a status update and yet to verify the output.**
>
> @klintan : I gave one Sunday to it.
> Converted it to ROS1. Getting Odom. However, I am still to verify the output w.r.t. the video and the [...]

Oh nice catch, that's even better then. I'll update my code, or if you want to feel free to do a PR on my branch and i'll approve.

rahuldeo2047, 2020-04-06:
@klintan : I tried the odom in ros1. However, I think I am back to camera calibration now. So that accuracy can be checked.
Without calibration I think it is invalid to comment on the accuracy of odom.

I tried with simple gazebo simulated various cameras but the calibrated output is not good enough. 
Had to use wide angle camera in gazebo for not losing track with turning sharp around 90degrees.
Any set up of camera is giving around 10cm displacement on 1m. After a meter the odom doesn't change. Tricky. 
However, I used my mobile phone and calibrated. The graph visualisation is appreciable for straight move and slight turns. Still the displacement in odom is not good.

I am back to camera calibration.

Thank you for the code.

rahuldeo2047, 2020-04-08:
**Status**: Gazebo custom wide-angle camera of 2.PI FOV, calibrated, all applications started without any issue. The viewer is not showing any image though. However, the openvslam as per logs seems to work. The Odom is still the same.

Still to try out camera calibration with the various simulated camera.

Zumbalamambo, 2020-05-15:
@rahuldeo2047 

It gives me the following matrix,

```

  0.923833  -0.115088  -0.365086  -0.143865
-0.0427792    -0.9788   0.200302  0.0126258
  0.380399   0.169427   0.909171   -0.10293
        -0          0         -0         -1

```

1. How do I interpret them? Where is the rotation matrix, xyz param, orientation of the camera?

if my understanding is right, `[-0.143865 0.0126258 -0.10293]` are the xyz positions and the rotation matrix is represented as,

```
> 0.923833 -0.115088  -0.365086 
> -0.0427792   -0.9788 0.200302
> 0.380399  0.169427  0.909171 
```

2. Is this matrix relative to the world? or do I have to calculate the world matrix [again](https://github.com/xdspacelab/openvslam/blob/2404c379f44d4ea252ebafc98d35b916f7bfbfe5/viewer/public/js/ViewControls.js#L157) with these values?

rahuldeo2047, 2020-05-15:
@Zumbalamambo : thank you for the info.
I am currently working on simulated cameras before buying one.
I am also looking for an cost effective solution. As it is my hobby. :)

Zumbalamambo, 2020-05-15:
@rahuldeo2047 what do you mean by simulated camera?

rahuldeo2047, 2020-05-15:
Using ros gazebo.

Zumbalamambo, 2020-05-15:
@rahuldeo2047 how much accuracy are you getting with simulated camera?

rahuldeo2047, 2020-05-15:
Invalid numbers as all zero.

I need to check on calibration as I am using non standard simulated cameras.

Shortcut will be to use the authors camera.

Zumbalamambo, 2020-05-15:
@rahuldeo2047 Could you point me a source where I can learn this?

rahuldeo2047, 2020-05-15:
I will check and put a link here.

nfynt, 2020-07-01:
> @rahuldeo2047
> 
> It gives me the following matrix,
> 
> ```
> 
>   0.923833  -0.115088  -0.365086  -0.143865
> -0.0427792    -0.9788   0.200302  0.0126258
>   0.380399   0.169427   0.909171   -0.10293
>         -0          0         -0         -1
> ```
> 
> 1. How do I interpret them? Where is the rotation matrix, xyz param, orientation of the camera?
> 
> if my understanding is right, `[-0.143865 0.0126258 -0.10293]` are the xyz positions and the rotation matrix is represented as,
> 
> ```
> > 0.923833 -0.115088  -0.365086 
> > -0.0427792   -0.9788 0.200302
> > 0.380399  0.169427  0.909171 
> ```
> 
> 1. Is this matrix relative to the world? or do I have to calculate the world matrix [again](https://github.com/xdspacelab/openvslam/blob/2404c379f44d4ea252ebafc98d35b916f7bfbfe5/viewer/public/js/ViewControls.js#L157) with these values?

What's the answer to this?

jediofgever, 2020-12-30:
how do you deal with the scale of map if you use wide angle cam. I am also using gazebo simulated cameras. RGBD cannot cope wit rotations. But at least the scale is OK. And now if I switch to wide angle cam, the scale will be the issue. I really wishthat there will be IMU integration to this...