xdspacelab/openvslam #161: better to undistort image before extract descriptor when using fisheye or equirectangular camera?

CanCanZeng, 2019-11-01:
I am wondering that is it better to undistort image before extract feature descriptor using fisheye or equirectangular camera. Because these kinds of camera has large distortion, the descriptor extracted from the distorted image is very inacurate. This will cause low feature match ratio when viewing from different direction, or even the same direction but from different position, since different position on the image has different distortion effect.
I know it is very slow to undistort whole image, maybe we can calculate the point pair location when computing ORB descriptor?

shinsumicco, 2019-11-05:
@CanCanZeng 
I think what you are saying is correct.
But, we currently don't undistort input images directory because its process needs more time than undistorting only keypoints.
I think one solution is to use GPU for image undistortion.
Do you have any ideas?

CanCanZeng, 2019-11-05:
I think maybe we can just undistort keypoints, but when computing the descriptors, we also undistort the  point pairs around the keypoints in `orb_extractor::compute_orb_descriptor`.
https://github.com/xdspacelab/openvslam/blob/8b5d9f95057792e8cb0525d1c21c4c8de16717ab/src/openvslam/feature/orb_extractor.cc#L676
we can get the undistorted pixel value via remap matrix which can be computed at the beginning and only once.

frspring, 2020-04-13:
There are papers indicating that descriptors from undistorted image don't make better match ratio(even lower).

CanCanZeng, 2020-04-14:
Hi @frspring  which paper do you mention? That result sounds interesting.

frspring, 2020-04-20:
here you can find it. 
https://www.researchgate.net/publication/335573245_Monocular_Visual_Odometry_Using_Fisheye_Lens_Cameras