xdspacelab/openvslam #162: How to configure system to get pure odometry?

olivebardoza, 2019-11-02:
Hi,

This is really cool and interesting. I'm interested in using openvslam for visual odometry. I would like to turn off loop closure and mapping, I just want to use it as odometry source. Is that possible?

shinsumicco, 2019-11-05:
@olivebardoza 
Thank you for your interest in this project.

You can disable the loop detection with `disable_loop_detector()` API.
https://github.com/xdspacelab/openvslam/blob/master/src/openvslam/system.h#L90

In addition, you may need to disable optimization of camera poses in the mapping module to fix previous camera poses.

rahuldeo2047, 2020-04-06:
For info and better issue tracking https://github.com/xdspacelab/openvslam/issues/157