xdspacelab/openvslam #163: libopencv_dnn.so.3.4 error while running.

furkancaglayan, 2019-11-04:
When I try to run any of the executables I get this error:

(myenv) name@name:~/Desktop/openvslam$ ./build/run_camera_slam 
./build/run_camera_slam: error while loading shared libraries: libopencv_dnn.so.3.4: cannot open shared object file: No such file or directory.

I tried [this](https://github.com/cggos/dip_cvqt/issues/1) but it didn't work. Any help is appreciated.

shinsumicco, 2019-11-05:
@furkancaglayan 
Did you try reinstalling OpenCV?
It seems that libopencv_dnn.so.3.4 is not placed at the appropriate directory.

furkancaglayan, 2019-11-05:
> @furkancaglayan
> Did you try reinstalling OpenCV?
> It seems that libopencv_dnn.so.3.4 is not placed at the appropriate directory.

Hi, yes I tried it. Also, I'm using conda opencv version 3.4.7.

shinsumicco, 2019-11-06:
Is `libopencv_dnn.so.3.4` included in OpenCV which installed via `conda`?
If not included, please try installing OpenCV via `apt` (for Ubuntu) or `brew` (for macOS).

furkancaglayan, 2019-11-06:
Yes, when I run 'sudo find / -name "libopencv_core.so.3.4*"' it gives me this:
/home/fcaglayan/anaconda3/envs/myenv/lib/libopencv_core.so.3.4
/home/fcaglayan/anaconda3/envs/myenv/lib/libopencv_core.so.3.4.7
/home/fcaglayan/anaconda3/pkgs/libopencv-3.4.2-hb342d67_1/lib/libopencv_core.so.3.4
/home/fcaglayan/anaconda3/pkgs/libopencv-3.4.2-hb342d67_1/lib/libopencv_core.so.3.4.2
/home/fcaglayan/anaconda3/pkgs/opencv-3.4.7-py37h8949d38_2/lib/libopencv_core.so.3.4
/home/fcaglayan/anaconda3/pkgs/opencv-3.4.7-py37h8949d38_2/lib/libopencv_core.so.3.4.7