xdspacelab/openvslam #167: Pangolin viewer not showing point cloud

borcherd, 2019-11-08:
When I try to run openvslam on my own video, filmed with a kodak pixpro sp360, it shows points on the video, but doesnt show anything in pangolin. 
The tutorial i used was https://openvslam.readthedocs.io/en/master/docker.html#instructions-for-pangolinviewer . 
Any help? :(

shinsumicco, 2019-11-08:
@borcherd 
Could you post a log shown in your terminal?

DeCatNiels, 2019-11-09:
@shinsumicco 
Hi, I'm working together with @borcherd on a project.
First we tried to use a video from a smartphone. This worked, although it lost its points of interest quite easily.
So now we wanted to test it with this fisheye camera (kodak pixpro sp360) to see if it could get enough data so it wouldn't lose the points of interest so easily.

To be complete we used the sample config of the LUMIX fisheye. And edited only the fps, cols and rows.

`# LUMIX fisheye model for 1K setting

#==============#
# Camera Model #
#==============#

Camera.name: "LUMIX DMC-GX8 8mm fisheye 1K"
Camera.setup: "monocular"
Camera.model: "fisheye"

Camera.fx: 441.730011169
Camera.fy: 442.520822476
Camera.cx: 480.35667528
Camera.cy: 275.490228646

Camera.k1: -3.068701296607466433e-02
Camera.k2: -3.343454364086094217e-03
Camera.k3: -2.881735840896060968e-03
Camera.k4: -5.917420310474077278e-04

Camera.fps: 30.0
Camera.cols: 2880
Camera.rows: 2880

Camera.color_order: "RGB"

#================#
# orb parameters #
#================#

Feature.max_num_keypoints: 3000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 5

#========================#
# Initializer Parameters #
#========================#

Initializer.scaling_factor: 5.0`

And this is the output we get from running following command in the openvslam container: `./run_video_slam -v ./orb_vocab/orb_vocab.dbow2 -m ./mediaan_office/fisheyepixpro.mp4 -c ./mediaan_office/config.yaml --frame-skip 3 --no-sleep --map-db map.msg --debug --eval-log`  

output log terminal:
<pre>root@e3990b15e0be:/openvslam/build# ./run_video_slam -v ./orb_vocab/orb_vocab.dbow2 -m ./mediaan_office/fisheyepixpro.mp4 -c ./mediaan_office/config.yaml --frame-skip 3 --no-sleep --map-db map.msg --debug --eval-log
[2019-11-09 09:12:38.568] <font color="#06989A">[D] CONSTRUCT: config</font>
[2019-11-09 09:12:38.568] <font color="#4E9A06">[I] config file loaded: ./mediaan_office/config.yaml</font>
[2019-11-09 09:12:38.568] <font color="#06989A">[D] load camera model type</font>
[2019-11-09 09:12:38.568] <font color="#06989A">[D] load camera model parameters</font>
[2019-11-09 09:12:38.568] <font color="#06989A">[D] CONSTRUCT: camera::base</font>
[2019-11-09 09:12:38.568] <font color="#06989A">[D] CONSTRUCT: camera::fisheye</font>
[2019-11-09 09:12:38.568] <font color="#06989A">[D] compute image bounds</font>
[2019-11-09 09:12:38.569] <font color="#06989A">[D] load ORB parameters</font>
[2019-11-09 09:12:38.569] <font color="#06989A">[D] load tracking parameters</font>
[2019-11-09 09:12:38.569] <font color="#06989A">[D] load depth threshold</font>
[2019-11-09 09:12:38.569] <font color="#06989A">[D] load depthmap factor</font>
[2019-11-09 09:12:38.569] <font color="#06989A">[D] CONSTRUCT: system</font>
  ___               __   _____ _      _   __  __ 
 / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
| (_) | &apos;_ \/ -_) &apos; \\ V /\__ \ |__ / _ \| |\/| |
 \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
      |_|                                        
[... 99 lines cut ...]
[2019-11-09 09:13:10.044] <font color="#4E9A06">[I] encoding 0 landmarks to store</font>
[2019-11-09 09:13:10.049] <font color="#4E9A06">[I] save the MessagePack file of database to map.msg</font>
median tracking time: 0.253368[s]
mean tracking time: 0.259671[s]
[2019-11-09 09:13:10.053] <font color="#06989A">[D] DESTRUCT: global_optimization_module</font>
[2019-11-09 09:13:10.053] <font color="#06989A">[D] DESTRUCT: mapping_module</font>
[2019-11-09 09:13:10.053] <font color="#06989A">[D] DESTRUCT: tracking_module</font>
[2019-11-09 09:13:10.053] <font color="#06989A">[D] DESTRUCT: module::relocalizer</font>
[2019-11-09 09:13:10.053] <font color="#06989A">[D] DESTRUCT: module::initializer</font>
[2019-11-09 09:13:10.053] <font color="#06989A">[D] DESTRUCT: initialize::perspective</font>
[2019-11-09 09:13:10.053] <font color="#4E9A06">[I] clear BoW database</font>
[2019-11-09 09:13:10.053] <font color="#06989A">[D] DESTRUCT: data::bow_database</font>
[2019-11-09 09:13:10.053] <font color="#4E9A06">[I] clear map database</font>
[2019-11-09 09:13:10.053] <font color="#06989A">[D] DESTRUCT: data::map_database</font>
[2019-11-09 09:13:10.053] <font color="#06989A">[D] DESTRUCT: data::camera_database</font>
[2019-11-09 09:13:10.133] <font color="#06989A">[D] DESTRUCT: system</font>
[2019-11-09 09:13:10.133] <font color="#06989A">[D] DESTRUCT: publish::map_publisher</font>
[2019-11-09 09:13:10.133] <font color="#06989A">[D] DESTRUCT: publish::frame_publisher</font>
[2019-11-09 09:13:10.135] <font color="#06989A">[D] DESTRUCT: camera::fisheye</font>
[2019-11-09 09:13:10.135] <font color="#06989A">[D] DESTRUCT: camera::base</font>
[2019-11-09 09:13:10.135] <font color="#06989A">[D] DESTRUCT: config</font>
</pre>

I hope you can get some info out of this.
Also thank you for the fast reply on the issue.

shinsumicco, 2019-11-11:
@DeCatNiels
Thank you for providing me the log.
Could you try the following steps?

- try using the latest `develop` branch (the fix for super-wide fisheye cameras has been included today with #165 by @matsuren )
- resize input images to half (please reconfigure the configuration file accordingly)

DeCatNiels, 2019-11-13:
@shinsumicco 
Thank for the help.
We just tried to use the develop branch. But when we run the video_slam command with the example of 'aist_living_lab' we get an a error. But we cant realy make something of it to be honest.

Here is the log of the terminal when we run it:
<pre>root@45701fc81b09:/openvslam/build# ./run_video_slam -v ./orb_vocab/orb_vocab.dbow2 -m ./aist_living_lab_1/video.mp4 -c ./aist_living_lab_1/config.yaml --frame-skip 3 --no-sleep --map-db map.msg
[2019-11-13 13:26:57.447] <font color="#4E9A06">[I] config file loaded: ./aist_living_lab_1/config.yaml</font>
  ___               __   _____ _      _   __  __ 
 / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
| (_) | &apos;_ \/ -_) &apos; \\ V /\__ \ |__ / _ \| |\/| |
 \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
      |_|                                        

Copyright (C) 2019,
National Institute of Advanced Industrial Science and Technology (AIST)
All rights reserved.

This is free software,
and you are welcome to redistribute it under certain conditions.
See the LICENSE file.

Camera Configuration:
- name: RICOH THETA S 960
- setup: Monocular
- fps: 30
- cols: 1920
- rows: 960
- color: RGB
- model: Equirectangular
ORB Configuration:
- number of keypoints: 2000
- scale factor: 1.2
- number of levels: 8
- initial fast threshold: 20
- minimum fast threshold: 7
- mask rectangles:
  - [0, 1, 0, 0.1]
  - [0, 1, 0.84, 1]
  - [0, 0.2, 0.7, 1]
  - [0.8, 1, 0.7, 1]
Tracking Configuration:

[2019-11-13 13:26:57.448] <font color="#4E9A06">[I] loading ORB vocabulary: ./orb_vocab/orb_vocab.dbow2</font>
terminate called after throwing an instance of &apos;std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt;&apos;
*** Aborted at 1573651617 (unix time) try &quot;date -d @1573651617&quot; if you are using GNU date ***
PC: @     0x7f896c8c1e97 gsignal
*** SIGABRT (@0x22) received by PID 34 (TID 0x7f897079d100) from PID 34; stack trace: ***
    @     0x7f896faee890 (unknown)
    @     0x7f896c8c1e97 gsignal
    @     0x7f896c8c3801 abort
    @     0x7f896d142957 (unknown)
    @     0x7f896d148ab6 (unknown)
    @     0x7f896d148af1 std::terminate()
    @     0x7f896d148d24 __cxa_throw
    @     0x7f896fd9eadb DBoW2::TemplatedVocabulary&lt;&gt;::loadFromBinaryFile()
    @     0x7f896fd9bec8 openvslam::system::system()
    @     0x55cca21b0956 (unknown)
    @     0x55cca21af44a (unknown)
    @     0x7f896c8a4b97 __libc_start_main
    @     0x55cca21af96a (unknown)
Aborted (core dumped)
root@45701fc81b09:/openvslam/build# 
</pre>

Or is this branch on the moment only working with wide-angle fisheye?