xdspacelab/openvslam #176: Possible removal of loop edge keyframe

riematrix, 2019-11-19:
Hi,
I found that the keyframe of loop edge could be removed from graph. I dought this is a bug for further optimization? The flag cannot_be_erased_ in keyframe is set in optimization process but never used.

I've noticed that there was a commit once but not merged into master branch.
See:
https://github.com/xdspacelab/openvslam/commit/9956fe7ceee1bec5e5ebb82bf2ffa6bfc0a6e33f

shinsumicco, 2019-11-20:
[Commit 9956fe7](https://github.com/xdspacelab/openvslam/commit/9956fe7ceee1bec5e5ebb82bf2ffa6bfc0a6e33f) has already been merged into `master` branch, hasn't it?

riematrix, 2019-11-20:
@shinsumicco Sorry for not checking out the modification. It is merged to master branch but the file index on github is old. Thanks!

closed 2019-11-20 by riematrix