xdspacelab/openvslam #177: encoding 0 keyframes to store

chengwei-peng, 2019-11-19:
```
root@Linux:/# run_video_slam -v /mnt/UDISK/orb_vocab.dbow2 -c /mnt/UDISK/con
fig.yaml -m /mnt/UDISK/video.mp4 --frame-skip 3 --map-db aist_living_lab_1_map.m
sg
[1970-01-01 09:29:28.767] [I] config file loaded: /mnt/UDISK/config.yaml
  ___               __   _____ _      _   __  __ 
 / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
| (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
 \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
      |_|                                        

Copyright (C) 2019,
National Institute of Advanced Industrial Science and Technology (AIST)
All rights reserved.

This is free software,
and you are welcome to redistribute it under certain conditions.
See the LICENSE file.

Camera Configuration:
- name: LUMIX DMC-GX8 8mm fisheye 1K
- setup: Monocular
- fps: 30
- cols: 960
- rows: 540
- color: RGB
- model: Fisheye
  - fx: 441.73
  - fy: 442.521
  - cx: 480.357
  - cy: 275.49
  - k1: -0.030687
  - k2: -0.00334345
  - k3: -0.00288174
  - k4: -0.000591742
  - min x: -1688.94
  - max x: 2616.09
  - min y: -968.628
  - max y: 1343.02
ORB Configuration:
- number of keypoints: 2000
- scale factor: 1.2
- number of levels: 8
- initial fast threshold: 20
- minimum fast threshold: 7
Tracking Configuration:

[1970-01-01 09:29:28.770] [I] loading ORB vocabulary: /mnt/UDISK/orb_vocab.dbow2
[1970-01-01 09:29:31.580] [I] startup SLAM system
[1970-01-01 09:29:31.581] [I] start mapping module
[1970-01-01 09:29:31.581] [I] start global optimization module
[1970-01-01 09:29:31.636] [I] terminate mapping module
[1970-01-01 09:29:31.637] [I] terminate global optimization module
[1970-01-01 09:29:31.638] [I] shutdown SLAM system
[1970-01-01 09:29:31.639] [I] encoding 1 camera(s) to store
[1970-01-01 09:29:31.639] [I] encoding 0 keyframes to store
[1970-01-01 09:29:31.639] [I] encoding 0 landmarks to store
[1970-01-01 09:29:31.646] [I] save the MessagePack file of database to aist_living_lab_1_map.msg
median tracking time: 1.75e-06[s]
mean tracking time: 1.75e-06[s]
[1970-01-01 09:29:31.649] [I] clear BoW database
[1970-01-01 09:29:31.649] [I] clear map database
root@Linux:/# 
```
What's happening here？

shinsumicco, 2019-11-20:
It seems that the video you use doesn't contain any valid images/frames.

chengwei-peng, 2019-11-22:
ok。In addition to the Pangolin display, can it be displayed in other ways?

closed 2019-11-28 by chengwei-peng