xdspacelab/openvslam #178: Installation for Windows 10

viet2411, 2019-11-19:
Hello everyone,
Can anyone show me step by step of installing this framework on Windows 10? I had read the installation document but still can not figure out how to install this correctly
Thank you and best regards.

vwvw, 2019-11-22:
Where are you stuck at? Are you running Docker or WSL?

saweer, 2019-11-23:
@viet2411 
I found this article: https://qiita.com/na0ki_ikeda/items/2310dc8dd70713de98e1
It is written in Japanese, but I was able to understand it after translating to English with Google Translate.

valeriob, 2020-02-18:
Hi @viet2411 
i was able to compile it for windows thanks to this repo https://github.com/reunanen/openvslam-build-vs it saved me sooo much time !

shinsumicco, 2020-04-26:
@viet2411 
Would you try this?
https://github.com/shinsumicco/openvslam-on-msvc/

closed 2020-04-26 by shinsumicco