xdspacelab/openvslam #186: multiple cameras?

gb2111, 2019-12-04:
Hello.
Does the project advise or can be converted to the case where we use multiple cameras with one PC in one location? So they practically share same database but track independent.
Thanks.

vwvw, 2019-12-04:
You can use stereo camera. I don't think there is yet support for >2 cameras.

gb2111, 2019-12-04:
thanks for reply. 
stereo camera is still one tracking object. i am thinking about something like running few single cameras and track them independently.

shinsumicco, 2019-12-05:
@gb2111 Do you mean "multiple camera model" like [this video](https://www.youtube.com/watch?v=mDe8MDa-Qm0)?

gb2111, 2019-12-06:
they say multi camera cluster. so this is still one tracking entity.
i mean like you have for example two cameras each going into arbitrary position but still building same map.
in theory I could have 2 instances of slam and each build own map but that would be waste of resources.

CameronFraser, 2020-04-30:
@gb2111 Did you ever figure that out? I'm wondering the same thing, is there a way to use multiple video files basically to build the same map?

slovak194, 2020-07-20:
> @gb2111 Do you mean "multiple camera model" like [this video](https://www.youtube.com/watch?v=mDe8MDa-Qm0)?

That's might be very useful!