xdspacelab/openvslam #191: Camera Position

OliHas, 2019-12-06:
Maybe it's a trivial question: 
How can I get the pose of every frame when running the ROS version?
I couldn't find a published topic... Is there a way to publish the pose as a `pose.msg`?

Thanks in advance!

klintan, 2019-12-07:
I'm looking for this as well. As of now it's basically a sink, no published topics at all. I'll start some work for ROS2 to also include published topics for odometry and maybe map markers as well.

pjende, 2020-01-20:
Any news on this? Would be also interested!

rahuldeo2047, 2020-01-27:
Thank you @klintan for https://github.com/xdspacelab/openvslam/pull/225
I'll check it ASAP.

rahuldeo2047, 2020-04-06:
For future ref only: https://github.com/xdspacelab/openvslam/issues/157