xdspacelab/openvslam #200: Stuck at "Initialization succeeded with E"

GokulEpiphany, 2019-12-12:
Thanks for this versatile framework. 
Followed through the simple tutorial:
PangolinViewer (both Map and Frame) opens up but no mapping occurs.
The only difference from the installation is I'm using opencv4.x and not Opencv3.4.0 (Could this be the issue?)
Also, I'm running this on jetson nano.

Fyi,
[2019-12-12 19:45:36.156] [I] config file loaded: aist_living_lab_1/config.yaml
  ___               __   _____ _      _   __  __ 
 / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
| (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
 \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
      |_|                                        

Copyright (C) 2019,
National Institute of Advanced Industrial Science and Technology (AIST)
All rights reserved.

This is free software,
and you are welcome to redistribute it under certain conditions.
See the LICENSE file.

Camera Configuration:
- name: RICOH THETA S 960
- setup: Monocular
- fps: 30
- cols: 1920
- rows: 960
- color: RGB
- model: Equirectangular
ORB Configuration:
- number of keypoints: 2000
- scale factor: 1.2
- number of levels: 8
- initial fast threshold: 20
- minimum fast threshold: 7
- mask rectangles:
  - [0, 1, 0, 0.1]
  - [0, 1, 0.84, 1]
  - [0, 0.2, 0.7, 1]
  - [0.8, 1, 0.7, 1]

[2019-12-12 19:45:36.157] [I] loading ORB vocabulary: orb_vocab/orb_vocab.dbow2
[2019-12-12 19:45:37.522] [I] startup SLAM system
[2019-12-12 19:45:37.522] [I] start mapping module
[2019-12-12 19:45:37.522] [I] start global optimization module
[2019-12-12 19:45:38.580] [I] initialization succeeded with E

Please let me know if you have any ideas. It would be useful in debugging.

shinsumicco, 2019-12-12:
@GokulEpiphany 
Could you post a log which is printed with `--debug` option?
After `initialization succeeded with E`, does the program terminate abnormally (e.g. SIGSEGV, SIGILL) or stuck before termination?

GokulEpiphany, 2019-12-14:
Thanks for looking into it.
The program does not terminate but nothing happens.
I'm running the following command:
./run_video_slam  -v orb_vocab/orb_vocab.dbow2 -c aist_living_lab_1/config.yaml -m aist_living_lab_1/video.mp4 --frame-skip 10 --map-db aist_living_lab_1_map.msg --debug

[2019-12-14 17:31:02.284] [D] CONSTRUCT: config
[2019-12-14 17:31:02.284] [I] config file loaded: aist_living_lab_1/config.yaml
[2019-12-14 17:31:02.284] [D] load camera model type
[2019-12-14 17:31:02.285] [D] load camera model parameters
[2019-12-14 17:31:02.285] [D] CONSTRUCT: camera::base
[2019-12-14 17:31:02.285] [D] CONSTRUCT: camera::equirectangular
[2019-12-14 17:31:02.285] [D] compute image bounds
[2019-12-14 17:31:02.285] [D] load ORB parameters
[2019-12-14 17:31:02.285] [D] load tracking parameters
[2019-12-14 17:31:02.286] [D] load depth threshold
[2019-12-14 17:31:02.286] [D] load depthmap factor
[2019-12-14 17:31:02.286] [D] CONSTRUCT: system
  ___               __   _____ _      _   __  __ 
 / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
| (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
 \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
      |_|                                        

Copyright (C) 2019,
National Institute of Advanced Industrial Science and Technology (AIST)
All rights reserved.

This is free software,
and you are welcome to redistribute it under certain conditions.
See the LICENSE file.

Camera Configuration:
- name: RICOH THETA S 960
- setup: Monocular
- fps: 30
- cols: 1920
- rows: 960
- color: RGB
- model: Equirectangular
ORB Configuration:
- number of keypoints: 2000
- scale factor: 1.2
- number of levels: 8
- initial fast threshold: 20
- minimum fast threshold: 7
- mask rectangles:
  - [0, 1, 0, 0.1]
  - [0, 1, 0.84, 1]
  - [0, 0.2, 0.7, 1]
  - [0.8, 1, 0.7, 1]

[2019-12-14 17:31:02.286] [I] loading ORB vocabulary: orb_vocab/orb_vocab.dbow2
[2019-12-14 17:31:03.673] [D] CONSTRUCT: data::camera_database
[2019-12-14 17:31:03.673] [D] CONSTRUCT: data::map_database
[2019-12-14 17:31:03.673] [D] CONSTRUCT: data::bow_database
[2019-12-14 17:31:03.693] [D] CONSTRUCT: publish::frame_publisher
[2019-12-14 17:31:03.693] [D] CONSTRUCT: publish::map_publisher
[2019-12-14 17:31:03.693] [D] CONSTRUCT: module::initializer
[2019-12-14 17:31:03.693] [D] CONSTRUCT: module::relocalizer
[2019-12-14 17:31:03.693] [D] CONSTRUCT: tracking_module
[2019-12-14 17:31:03.693] [D] CONSTRUCT: mapping_module
[2019-12-14 17:31:03.693] [D] CONSTRUCT: global_optimization_module
[2019-12-14 17:31:03.693] [I] startup SLAM system
[2019-12-14 17:31:03.694] [I] start mapping module
[2019-12-14 17:31:03.694] [I] start global optimization module
[2019-12-14 17:31:04.250] [D] CONSTRUCT: initialize::bearing_vector
[2019-12-14 17:31:04.960] [I] initialization succeeded with E
[2019-12-14 17:31:04.960] [D] DESTRUCT: initialize::bearing_vector

shinsumicco, 2019-12-20:
Hmm... It seems an issue related to optimization with g2o, but I could not find out the cause for the log you posted.
We continuously inspect it.

What version (commit hash) of g2o are you using for OpenVSLAM?

GokulEpiphany, 2019-12-20:
Using the same version as mentioned in instructions:
git checkout 9b41a4ea5ade8e1250b9c1b279f3a9c098811b5a

Do let me know if anything pops up. Thanks for the reply.

fhirmann, 2019-12-30:
Hi,

It seems like I have a similar issue.

I installed OpenVSLAM as described at https://openvslam.readthedocs.io/en/master/installation.html and tried the very basic tutorial from https://openvslam.readthedocs.io/en/master/simple_tutorial.html. However, I also get stuck at the very beginning.

My system is Ubuntu 18.04. I also tried to build it with -DBUILD_WITH_MARCH_NATIVE=OFF but it's the same result.

My output (with --debug) is:

> ./run_video_slam -v ./orb_vocab/orb_vocab.dbow2 -m ./aist_living_lab_1/video.mp4 -c 
> ./aist_living_lab_1/config.yaml --frame-skip 3 --no-sleep --map-db map.msg --debug
> [2019-12-30 17:40:37.394] [D] CONSTRUCT: config
> [2019-12-30 17:40:37.394] [I] config file loaded: ./aist_living_lab_1/config.yaml
> [2019-12-30 17:40:37.394] [D] load camera model type
> [2019-12-30 17:40:37.394] [D] load camera model parameters
> [2019-12-30 17:40:37.394] [D] CONSTRUCT: camera::base
> [2019-12-30 17:40:37.394] [D] CONSTRUCT: camera::equirectangular
> [2019-12-30 17:40:37.394] [D] compute image bounds
> [2019-12-30 17:40:37.394] [D] load ORB parameters
> [2019-12-30 17:40:37.395] [D] load tracking parameters
> [2019-12-30 17:40:37.395] [D] load depth threshold
> [2019-12-30 17:40:37.395] [D] load depthmap factor
> [2019-12-30 17:40:37.395] [D] CONSTRUCT: system
>   ___               __   _____ _      _   __  __ 
>  / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
> | (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
>  \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
>       |_|                                        
> 
> Copyright (C) 2019,
> National Institute of Advanced Industrial Science and Technology (AIST)
> All rights reserved.
> 
> This is free software,
> and you are welcome to redistribute it under certain conditions.
> See the LICENSE file.
> 
> Camera Configuration:
> - name: RICOH THETA S 960
> - setup: Monocular
> - fps: 30
> - cols: 1920
> - rows: 960
> - color: RGB
> - model: Equirectangular
> ORB Configuration:
> - number of keypoints: 2000
> - scale factor: 1.2
> - number of levels: 8
> - initial fast threshold: 20
> - minimum fast threshold: 7
> - mask rectangles:
>   - [0, 1, 0, 0.1]
>   - [0, 1, 0.84, 1]
>   - [0, 0.2, 0.7, 1]
>   - [0.8, 1, 0.7, 1]
> 
> [2019-12-30 17:40:37.395] [I] loading ORB vocabulary: ./orb_vocab/orb_vocab.dbow2
> [2019-12-30 17:40:37.931] [D] CONSTRUCT: data::camera_database
> [2019-12-30 17:40:37.931] [D] CONSTRUCT: data::map_database
> [2019-12-30 17:40:37.931] [D] CONSTRUCT: data::bow_database
> [2019-12-30 17:40:37.932] [D] CONSTRUCT: publish::frame_publisher
> [2019-12-30 17:40:37.932] [D] CONSTRUCT: publish::map_publisher
> [2019-12-30 17:40:37.932] [D] CONSTRUCT: module::initializer
> [2019-12-30 17:40:37.932] [D] CONSTRUCT: module::relocalizer
> [2019-12-30 17:40:37.932] [D] CONSTRUCT: tracking_module
> [2019-12-30 17:40:37.932] [D] CONSTRUCT: mapping_module
> [2019-12-30 17:40:37.932] [D] CONSTRUCT: global_optimization_module
> [2019-12-30 17:40:37.932] [I] startup SLAM system
> [2019-12-30 17:40:37.932] [I] start mapping module
> [2019-12-30 17:40:37.932] [I] start global optimization module
> [2019-12-30 17:40:38.061] [D] CONSTRUCT: initialize::bearing_vector
> [2019-12-30 17:40:38.152] [I] initialization succeeded with E
> [2019-12-30 17:40:38.152] [D] DESTRUCT: initialize::bearing_vector

I appreciate any help.

Thanks,
Fabian

EliasBjorne, 2020-01-06:
I have the same issue. I also had problems when I compiled OpenCV from source and also having CUDA 10. I did the fix according to [opencv/issues/1786]( https://github.com/opencv/opencv_contrib/issues/1786)(cmake -DBUILD_opencv_cudacodec=OFF ...). I don't know if this has anything to do with the problem. Please let me know if anyone figures something out 

Thanks
Elias

Kawal15, 2020-02-12:
I encountered the same problem, here is what i discovered:
I tried to reinstall openvslam and it was suddenly working, I then continued to install the ROS package and now it didn't work (it was stuck after [I] initialization succeeded with E). I realized it was using an old version of OpenCV I had installed on my PC and not OpenCV 3.4.0 as I installed in the instructions.

> Please build OpenVSLAM with OpenCV 3.3.1 or later if you plan on using ROS package.

Using a version above 3.3.1 fixed the problem for me. 

Kasper

YujiElfahkrany, 2020-03-30:
I had the same issue and it was fixed after I removed ros-melodic/libg2o using synaptic

rahuldeo2047, 2020-04-06:
@YujiElfahkrany : same

seraaaayeo, 2020-05-02:
@YujiElfahkrany I fixed the problem. Thx

GokulEpiphany, 2020-05-05:
@YujiElfahkrany  Your solution seems to have worked. Thanks. Closing

closed 2020-05-05 by GokulEpiphany

rashikshrestha, 2020-08-06:
> I had the same issue and it was fixed after I removed ros-melodic-libg2o using synaptic

Thanks, it worked for me too :)