xdspacelab/openvslam #204: with socket running

henrywen2011, 2019-12-16:
please check the bug info,thanks!

[2019-12-16 18:44:57.951] [I] startup SLAM system
[2019-12-16 18:44:57.951] [I] start mapping module
[2019-12-16 18:44:57.951] [I] start global optimization module
[2019-12-16 18:44:57] [connect] Successful connection
[2019-12-16 18:44:57] [connect] WebSocket Connection 127.0.0.1:3000 v-2 "WebSocket++/0.8.1" /socket.io/?EIO=4&transport=websocket&t=1576493097 101
[2019-12-16 18:44:57.953] [I] connected to server
[2019-12-16 18:45:00.819] [I] initialization succeeded with F
double free or corruption (out)
*** Aborted at 1576493100 (unix time) try "date -d @1576493100" if you are using GNU date ***
PC: @     0x7fb9c69b1e97 gsignal
*** SIGABRT (@0x3e80000389d) received by PID 14493 (TID 0x7fb9b053c700) from PID 14493; stack trace: ***
    @     0x7fb9c924c890 (unknown)
    @     0x7fb9c69b1e97 gsignal
    @     0x7fb9c69b3801 abort
    @     0x7fb9c69fc897 (unknown)
    @     0x7fb9c6a0390a (unknown)
    @     0x7fb9c6a0ae75 cfree
    @     0x7fb9c1b59098 g2o::OptimizableGraph::~OptimizableGraph()
    @     0x7fb9c98a4082 openvslam::optimize::global_bundle_adjuster::optimize()
    @     0x7fb9c9851089 openvslam::module::initializer::create_map_for_monocular()
    @     0x7fb9c9851a73 openvslam::module::initializer::initialize()
    @     0x7fb9c97bfecc openvslam::tracking_module::initialize()
    @     0x7fb9c97c1db5 openvslam::tracking_module::track()
    @     0x7fb9c97c21f2 openvslam::tracking_module::track_monocular_image()
    @     0x7fb9c97b483c openvslam::system::feed_monocular_frame()
    @     0x560ccd2399cc (unknown)
    @     0x7fb9c70396ef (unknown)
    @     0x7fb9c92416db start_thread
    @     0x7fb9c6a9488f clone
Aborted (core dumped)

shinsumicco, 2019-12-20:
Could you configure `cmake` with `-DBUILD_WITH_MARCH_NATIVE=OFF` or  `-DBUILD_WITH_MARCH_NATIVE=ON` option?

safijari, 2019-12-22:
For any that might be in suspense: I just had this issue (was using `DBUILD_WITH_MARCH_NATIVE=ON`. Changed it to `OFF` and the segfault went away.

henrywen2011, 2019-12-23:
Thanks  shinsumicco, it's ok  change it to OFF.

closed 2019-12-23 by henrywen2011