xdspacelab/openvslam #205: Merging maps

EmreOzkose, 2019-12-16:
Hi! You mentioned a feature ("Created maps can be stored and loaded, then OpenVSLAM can localize new images based on the prebuilt maps.") in README. However, Does the framework contain merging maps which are created in different time?

Thank you.

Shashika007, 2019-12-20:
@EmreOzkose So far, they have not implemented map merging technique with openvslam like maplab.

shinsumicco, 2019-12-20:
@EmreOzkose 
Currently OpenVSLAM does not have any function to merge maps captured in different times.
It can be done by matching images between different maps via geometric verification (e.g. epipolar constraint) and estimate a relative transformation between the two maps like global SfM.