xdspacelab/openvslam #206: Getting camera position

Wrs-0, 2019-12-19:
Is there a way to get the camera's position while the map is being created? Which file should I be looking at that outputs the camera position while the program is running.

shinsumicco, 2019-12-20:
`system::feed_XXXX_frame` returns a camera pose corresponding to the feeded frame.
https://github.com/xdspacelab/openvslam/blob/d19af6ba1918be4db46db598b46beea8948bd76b/src/openvslam/system.h#L104-L114
Is it enough for your purpose?

Wrs-0, 2019-12-20:
This may seem like a stupid question, but by camera pose do you mean the camera's tilted position or the camera's position relative to the map. Thankyou for your time though.

jessearmand, 2019-12-27:
@shinsumicco I can see that the camera pose is a 4x4 matrix but it really is difficult to understand the meaning of the values in the matrix without some more explanation. 

I suppose here we can get the camera pose relative to the map?

https://github.com/xdspacelab/openvslam/blob/2411387e693c9cfd4d43a2b90c0a7054bbecf745/src/openvslam/publish/map_publisher.h#L48

 Is the camera pose structured in x, y, z and quaternion values for the orientation? If so, how are these values represented inside the matrix?

DanielArnett, 2020-01-03:
@jessearmand It looks like a transformation matrix. The top left 3x3 is the rotation matrix, and the x,y,z position is usually in the rightmost column. [Here's an online calculator where you can create rotation matrices from euler angles.](https://www.andre-gaschler.com/rotationconverter/)

[This explains how you create a rotation matrix.](https://math.stackexchange.com/questions/651413/given-the-degrees-to-rotate-around-axis-how-do-you-come-up-with-rotation-matrix) They take some getting used to, but once you're comfortable with them they're really nice. Just one standard way of expressing any given position and orientation.

jessearmand, 2020-01-06:
@DanielArnett I see, thanks.

closed 2020-01-10 by Wrs-0

rahuldeo2047, 2020-04-06:
Ref only : https://github.com/xdspacelab/openvslam/issues/157

Zumbalamambo, 2020-05-28:
@DanielArnett May I know what origin refers to? How do I get the origin in the map?

nfynt, 2020-07-01:
> @jessearmand It looks like a transformation matrix. The top left 3x3 is the rotation matrix, and the x,y,z position is usually in the rightmost column. [Here's an online calculator where you can create rotation matrices from euler angles.](https://www.andre-gaschler.com/rotationconverter/)
> 
> [This explains how you create a rotation matrix.](https://math.stackexchange.com/questions/651413/given-the-degrees-to-rotate-around-axis-how-do-you-come-up-with-rotation-matrix) They take some getting used to, but once you're comfortable with them they're really nice. Just one standard way of expressing any given position and orientation.
> 
> For example here's a transformation matrix that doesn't do anything. In other words it means the camera is just sitting still at the origin. (assume there are zeros where the table is blank)
> 
> 1			
> 1		
> 1	
> 1
> Here's one where the camera has moved 1 unit in the x and the z direction
> 
> 1			1
> 1		
> 1	1
> 1
> ### Matrix A
> Here's one with a 90 degree rotation around the x axis.
> 
> 1			
> cos(90)	-sin(90)	
> sin(90)	cos(90)	
> 1
> Which is the same as
> 
> 1			
> 0	-1	
> 1	0	
> 1
> ### Matrix B
> So finally, if you want the camera to go to the XYZ position (1, 0, 1) like in the above matrix, and then rotate 90 degrees about the X axis, you just multiply the two matrices above: AxB=C
> 
> 1			1
> -1	
> 1		1
> 1
> But beware! If you did the multiplication in the wrong order you'll end up rotating and then doing the movement after the rotation, ending up with
> 
> 1			1
> -1	-1
> 1		
> 1
> So if you just want the camera position you can just get the top 3 values in the right column, and that will be your x, y, z position.
> 
> If you want your orientation, you'll need to take the values in the top left 3x3 and convert that to euler angles or a quaternion, which you'll probably just want to find some library to do for you.

@DanielArnett 
Thanks for the explanation. However in my case when I measure the travel a distance of ~30m straight in the direction of camera (z-axis), the camera translation is ~4.xx units (translation in z-axis)! Is this due to ambiguity of scaling factor? If so how can I fix this, I have tried updating scale value manually from camera config file, but that doesn't work.

How can I get accurate camera motion in the real world?