xdspacelab/openvslam #207: Can i run SLAM without orb_vocab.dbow2? What will happen then?

SKioresku, 2019-12-19:


Shashika007, 2019-12-20:
@SKioresku You have to choose either dbow2 or fbow as your bag of word library. In slam, we mostly use that for relocalization and loop closure purposes. openvslam won't work without a vocabulary file.

SKioresku, 2019-12-20:
I wish to use openvslam for web application, but 40 mb is too much for downloading with mobile device. What can i do((?

shinsumicco, 2019-12-20:
@SKioresku 
I recommend that you create own vocabulary whose size is smaller than the one we provide.
Vocabulary size is mainly influenced by parameters for vocabulary creation.

SKioresku, 2019-12-20:
Can i use DBoW3 for vocabulary creation or i need to use DBoW2 vertion of shincumicco? What version is better and compatible with openvslam?

Shashika007, 2019-12-21:
@SKioresku you can use DBoW3, but in order to use that, you have to change code inside source also BoW loading function. Also, you can train your own BoW file eaither dbow2 or 3 . I suggest you to try FBoW which is already implemented inside openvslam.

SKioresku, 2019-12-21:
Can i resize training images to resolution performing for tracking ( 640x480 )?

SKioresku, 2019-12-23:
Thanks for the help)

closed 2019-12-23 by SKioresku

dzqiu, 2020-04-24:
Hello, I can't download Bovisa dataset from the torrent. Anyone can help me? Thanks.