xdspacelab/openvslam #208: Video SLAM with Ricoh Theta V not working

zainmehdi, 2019-12-20:
Hi,
First of all thanks alot for the great package. I tried to run Video slam using video recorded with Ricoh Theta V model. Here is my config file

# Equirectangular model for RICOH THETA S

#==============#
# Camera Model #
#==============#

Camera.name: "RICOH THETA V"
Camera.setup: "monocular"
Camera.model: "equirectangular"

Camera.fps: 30.0
Camera.cols: 3849
Camera.rows: 1920

Camera.color_order: "RGB"

#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 2000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7
Feature.mask_rectangles:
  - [0.0, 1.0, 0.0, 0.1]
  - [0.0, 1.0, 0.84, 1.0]
  - [0.0, 0.2, 0.7, 1.0]
  - [0.8, 1.0, 0.7, 1.0]

It seems that features are being detected and the app starts but I am unable to initialize mapping as the video goes on. There is no localization and mapping at all and pangolin viewer just shows a stationary keyframe. 

Here is the output 
Camera Configuration:
- name: RICOH THETA V
- setup: Monocular
- fps: 30
- cols: 3849
- rows: 1920
- color: RGB
- model: Equirectangular
ORB Configuration:
- number of keypoints: 2000
- scale factor: 1.2
- number of levels: 8
- initial fast threshold: 20
- minimum fast threshold: 7
- mask rectangles:
  - [0, 1, 0, 0.1]
  - [0, 1, 0.84, 1]
  - [0, 0.2, 0.7, 1]
  - [0.8, 1, 0.7, 1]

[2019-12-20 09:14:30.613] [I] loading ORB vocabulary: ./orb_vocab/orb_vocab.dbow2
[2019-12-20 09:14:31.066] [I] startup SLAM system
[2019-12-20 09:14:31.066] [I] start mapping module
[2019-12-20 09:14:31.066] [I] start global optimization module
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.

Would be thankful for any help.

zainmehdi, 2019-12-20:
Never mind got it. While recording the video a large portion of the field of view was being blocked by the recorder preventing initialization.

closed 2019-12-20 by zainmehdi

johnny04thomas, 2020-01-08:
@zainmehdi  Did you run the openvslam on a linux distribution? If so can tell me how you found a driver to capture video from Ricoh theta V.
Thanking you in advance.

zainmehdi, 2020-01-08:
> @zainmehdi Did you run the openvslam on a linux distribution? If so can tell me how you found a driver to capture video from Ricoh theta V.
> Thanking you in advance.

I actually recorded the video and then ran vslam on it. As of now there is no UVC 1.5 support on linux so in my knowledge cant stream live video from ricoh theta v

johnny04thomas, 2020-01-08:
Thank you for your reply. I actually need to work with theta Z1 and was searching for a UVC 1.5 compliant driver. I found a suggestion in https://sourceforge.net/p/linux-uvc/mailman/message/32291864/ to use guvcview, but have no luck so far. Theta Z1 is getting detected as a media file (/dev/media*) rather than a video file (/dev/video*).

zainmehdi, 2020-01-08:
@johnny04thomas I I have also been digging into RICOH theta's official forum for the past two days and seems like there is no working solution available. I opted for offline processing. But would be great to have real-time streaming.

duguowei1000, 2020-01-21:
Hi all, i intent to use RICOH theta for real-time slam. Is it still doable as long as it is UVC 1.1? RICOH THETA S ( 1280x720 @ 15fps | MotionJPEG | UVC 1.1) should work for live processing?

Thanks

zainmehdi, 2020-01-21:
> Hi all, i intent to use RICOH theta for real-time slam. Is it still doable as long as it is UVC 1.1? RICOH THETA S ( 1280x720 @ 15fps | MotionJPEG | UVC 1.1) should work for live processing?
> 
> Thanks

I think it should work. I havnt tried it yet myself , please update after your tests

duguowei1000, 2020-01-22:
But the resolution for theta S is much lower > will this affect ORB feature extraction much?

vlordier, 2020-05-24:
Maybe using ["SuperGlue"](https://arxiv.org/pdf/1911.11763.pdf) would help.