xdspacelab/openvslam #209: Error compiling ros for Openvslam (related to g2o)

duguowei1000, 2019-12-20:
Hi guys, i am not sure if this is g2o issue or openvslam. But just trying my luck here. i am trying to compile ros package in openvslam.
------------------------------------------------------------------------------------
Steps as per instruction in https://openvslam.readthedocs.io/en/master/ros_package.html:

cd /path/to/working/dir
git clone https://github.com/RainerKuemmerle/g2o.git
cd g2o
git checkout 9b41a4ea5ade8e1250b9c1b279f3a9c098811b5a
mkdir build && cd build
cmake \
    -DCMAKE_BUILD_TYPE=Release \
    -DCMAKE_INSTALL_PREFIX=/usr/local \
    -DCMAKE_CXX_FLAGS=-std=c++11 \
    -DBUILD_SHARED_LIBS=ON \
    -DBUILD_UNITTESTS=OFF \
    -DBUILD_WITH_MARCH_NATIVE=ON \
    -DG2O_USE_CHOLMOD=OFF \
    -DG2O_USE_CSPARSE=ON \
    -DG2O_USE_OPENGL=OFF \
    -DG2O_USE_OPENMP=ON \
    ..
make -j4
make install

-----------------------------------------------------------------------------
Error generated after "make install step":
...
[ 80%] Linking CXX executable /home/gw1604/vslamopen/ros/devel/lib/openvslam/run_slam
[ 86%] Linking CXX executable /home/gw1604/vslamopen/ros/devel/lib/openvslam/run_localization
/home/gw1604/vslamopen/build/lib/libopenvslam.so: undefined reference to `g2o::OptimizableGraph::addVertex(g2o::OptimizableGraph::Vertex*, g2o::HyperGraph::Data*)'
/home/gw1604/vslamopen/build/lib/libopenvslam.so: undefined reference to `g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(std::unique_ptr<g2o::Solver, std::default_delete<g2o::Solver> >)'
/home/gw1604/vslamopen/build/lib/libopenvslam.so: undefined reference to `g2o::OptimizableGraph::addEdge(g2o::OptimizableGraph::Edge*)'
collect2: error: ld returned 1 exit status
openvslam/src/CMakeFiles/run_slam.dir/build.make:155: recipe for target '/home/gw1604/vslamopen/ros/devel/lib/openvslam/run_slam' failed
make[2]: *** [/home/gw1604/vslamopen/ros/devel/lib/openvslam/run_slam] Error 1
/home/gw1604/vslamopen/build/lib/libopenvslam.so: undefined reference to `g2o::OptimizableGraph::addVertex(g2o::OptimizableGraph::Vertex*,CMakeFiles/Makefile2:2051: recipe for target 'openvslam/src/CMakeFiles/run_slam.dir/all' failed
 g2o::make[1]: *** [openvslam/src/CMakeFiles/run_slam.dir/all] Error 2
HyperGraph::Data*)'
/home/gw1604/vslamopen/make[1]: *** Waiting for unfinished jobs....
build/lib/libopenvslam.so: undefined reference to `g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(std::unique_ptr<g2o::Solver, std::default_delete<g2o::Solver> >)'
/home/gw1604/vslamopen/build/lib/libopenvslam.so: undefined reference to `g2o::OptimizableGraph::addEdge(g2o::OptimizableGraph::Edge*)'
collect2: error: ld returned 1 exit status
openvslam/src/CMakeFiles/run_localization.dir/build.make:155: recipe for target '/home/gw1604/vslamopen/ros/devel/lib/openvslam/run_localization' failed
make[2]: *** [/home/gw1604/vslamopen/ros/devel/lib/openvslam/run_localization] Error 1
CMakeFiles/Makefile2:1989: recipe for target 'openvslam/src/CMakeFiles/run_localization.dir/all' failed
make[1]: *** [openvslam/src/CMakeFiles/run_localization.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

------------------------------------------------------------------------------------------
Appreciate if any one has any idea( even just pointing out if it is g2o or Openvslam issue. Thanks!

Shashika007, 2019-12-20:
@duguowei1000 try the newer version of g2o without checking out 9b41a4ea5ade8e1250b9c1b279f3a9c098811b5a branch.

jungladicitta, 2020-01-21:
> @duguowei1000 try the newer version of g2o without checking out 9b41a4ea5ade8e1250b9c1b279f3a9c098811b5a branch.

I tried. It didnt help

jungladicitta, 2020-01-21:
@duguowei1000 Looks like I have similar issue. Have you solved it?

duguowei1000, 2020-01-29:
hi @jungladicitta , have you solved it?

jungladicitta, 2020-01-30:
have solved, but looks like there is another issue now, look up at the last comments there

Hsu-Hsin, 2020-08-13:
> @duguowei1000 have solved, but looks like there is another issue now, look up at the last comments there
How did you soleve it? I mean where are last comments that you said?

Hsu-Hsin, 2020-08-13:
How did you soleve it? I mean where are last comments that you said? @jungladicitta

Hsu-Hsin, 2020-08-13:
> @duguowei1000 have solved, but looks like there is another issue now, look up at the last comments there
How did you solve it? I mean where are last comments that you said? @jungladicitta