xdspacelab/openvslam #210: Tracking is perfect ... until it turns

sebastianpinedaar, 2019-12-20:
I am running the system for my own video. I already calibrated the camera accordingly. However, the system has a weird behavior: at the beginning, it works fine when the tracking is performed in a straight line, but when it turns left or right, the tracking is eventually lost. At other times, after turning, the viewer shows the current keyframe (green square) spinning around repeatedly although the video shows a following up straight path. Has somebody an idea what is happening? Can it be due to the fact that the video shows very close objects (less than one meter close)?

rahuldeo2047, 2020-03-01:
same with me and looking for help. Why it looses track while turning and at fast translations slower than 5km/h.

Shashika007, 2020-03-03:
@rahuldeo2047 @sebastianpinedaar  Normally, visual slam is not that robust to fast translations or turning behaviors. You can fuse IMU to get better tracking with faster movements ex : ([visual inertial slam](https://github.com/xdspacelab/openvslam/tree/feat/imu-integration)). Sometimes, the tracking performance depends on your CPU performance.

rahuldeo2047, 2020-03-03:
I think if we can have left right imags stitched, calibrated. Then I think turns can be taken care at speeds less than 5kmph.
However, for fast motions faster cpus and other sensor fusion can surely work.

Now how do we fuse multiple sensors.
As stitching and calibration is comparatively easy.

rahuldeo2047, 2020-04-06:
Ref only for future: https://github.com/xdspacelab/openvslam/issues/157

zlrpuck, 2020-04-14:
Would it help by using equirectangular cameras?

rahuldeo2047, 2020-04-14:
It should be verified, as I think it will track better in case of the mentioned camera. 

I think the authors used the same.

CameronFraser, 2020-05-14:
@rahuldeo2047 Could probably dig into this repo to figure out the sensor fusion https://github.com/HKUST-Aerial-Robotics/VINS-Fusion There is also a branch on the OpenVSLAM repo for integrating IMU data, but I'm not sure if development is still going on it or if the authors/maintainers got caught up with other projects. I've had much better luck with equirectangular cameras than perspective.

rahuldeo2047, 2020-05-14:
Thank you @CameronFraser I will look into it.