xdspacelab/openvslam #216: Invalid node error: this may result from using a map iterator as a sequence iterator, or vice-versa

squaresoft2015, 2020-01-02:
I ran the sample tutorial with the video sequence provided in Google drive. After the vocabulary, *.mp4 and *.yaml files are loaded, the program prompts a invalid node error. I have tried to figure out the reason by running the program with --debug option, here is the output:
(base) ubuntu@ubuntu-Macmini:~/openvslam-master/build$ ./run_video_slam -v orb_vocab/orb_vocab.dbow2 -c aist_entrance_hall_1/config.yaml -m aist_entrance_hall_2/video.mp4 --frame-skip 3 --map-db aist_entrance_hall_1_map.msg --debug
[2019-12-19 15:37:14.769] [D] CONSTRUCT: config
[2019-12-19 15:37:14.769] [I] config file loaded: aist_entrance_hall_1/config.yaml
[2019-12-19 15:37:14.769] [D] load camera model type
invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa

I have checked the files and they are all right. Can someone help?

MikiyaShibuya, 2020-01-08:
I think it caused by config.yaml.
I can check your config if you paste the file contents.

squaresoft2015, 2020-01-08:
The config.yaml file located in the directory of aist_entrance_hall_1, which is downloaded from Google drive:
# Equirectangular model for RICOH THETA S

#==============#
# Camera Model #
#==============#

Camera.name: "RICOH THETA S 960"
Camera.setup: "monocular"
Camera.model: "equirectangular"

Camera.fps: 30.0
Camera.cols: 1920
Camera.rows: 960

Camera.color_order: "RGB"

#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 2000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7
Feature.mask_rectangles:
  - [0.0, 1.0, 0.0, 0.1]
  - [0.0, 1.0, 0.84, 1.0]