xdspacelab/openvslam #218: Monocular slam specify scale

unrealwill, 2020-01-06:
Hello, 
How can I specify a real world measurement in order to provide a reference to resolve scale and coordinates frame ? (Feeding a stereo frame would do the trick but I'm doing monocular)

For example in Ucoslam one can set aruco_markerSize so that we provide an object size reference to solve the scale issue. 

Can something like that be implemented in openvslam, as it will also help fight scale drift.

For the coordinate position & orientation reference, we can always work in the frame reference of the first Keyframe, but it would be great if we could specify the position and orientation of a KeyFrame and freeze it as a reference, but it will probably induce some numerical instabilities to have a hard constraint.

To fight orientation and position drift, it would be great to be able to add compass from IMU, but it would be great to have some way to manually constrain some orientation of a KeyFrame, and position of a KeyFrame.

Thanks

pjende, 2020-01-20:
Hi, 

this is quite a late reply but what you are trying to achieve is not so simple since monocular setups are only up to scale. 

In order to retrieve the pose of the camera as well as the world points in a reference frame, a similarity transform could be performed. A similar project, OpenRealm (https://github.com/laxnpander/OpenREALM), does exactly what you require. Although it is based on ROS and other techniques, it is worth checking out. 

The issue with georeferencing the sparse points as well as the camera poses is outlined in this document on OpenRealm: https://drive.google.com/open?id=1zXxuCVN4wzlC-ZAZBz5qI4wLkv1b6CbH (page 36ff.)

A similar project following a slightly different approach can be found here https://github.com/hitimo/aerial_mapper

Zumbalamambo, 2020-06-07:
@sev-of-nine could you please suggest some methods to do similarity transforms with mono slam?