xdspacelab/openvslam #220: compute_stereo_from_depth issue code seems to  stop working

RahulSurana123, 2020-01-07:
*** Aborted at 1578404648 (unix time) try "date -d @1578404648" if you are using GNU date ***
PC: @     0x7f02a5163a1c openvslam::data::frame::compute_stereo_from_depth()
*** SIGSEGV (@0x7f0237fce638) received by PID 11943 (TID 0x7f02577fe700) from PID 939320888; stack trace: ***
    @     0x7f02a4e76390 (unknown)
    @     0x7f02a5163a1c openvslam::data::frame::compute_stereo_from_depth()
    @     0x7f02a5164bf2 openvslam::data::frame::frame()
    @     0x7f02a51347a1 openvslam::tracking_module::track_RGBD_image()
    @     0x7f02a5126474 openvslam::system::feed_RGBD_frame()
    @           0x41cc45 _ZZ13rgbd_trackingRKSt10shared_ptrIN9openvslam6configEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEjSC_fSC_ENKUlvE_clEv.constprop.849
    @     0x7f029e541c80 (unknown)
    @     0x7f02a4e6c6ba start_thread
    @     0x7f029dfb041d clone
    @                0x0 (unknown)
Segmentation fault (core dumped)

RahulSurana123, 2020-01-07:
i dont understand whats the problem going on here i am trying to take live frameset from realsense rgbd camera and use slam on it

Gilaine, 2020-01-08:
I have the same problem did you find any solution ?

RahulSurana123, 2020-01-09:
i am trying to debug the problem from the core i think its the error due to g2o file hence will be putting this error on there issue hence i am closing this issue here

closed 2020-01-09 by RahulSurana123

RahulSurana123, 2020-01-10:
i have got the solution to this problem

Gilaine, 2020-01-10:
How did you solve this problem?

RahulSurana123, 2020-01-13:
@Gilaine  have you found the solution to your problem ?

RahulSurana123, 2020-01-15:
ya i just matched the frame size using the config file if you need more help then please tell the camera module you are using...

shirbarzel, 2020-05-13:
> ya i just matched the frame size using the config file if you need more help then please tell the camera model you are using i was using realsense D435i ...

@RahulSurana123 
Can you share the config files you used to run it?