xdspacelab/openvslam #221: Run_video_slam.exe stop half way.

michealChin, 2020-01-10:
[image]

I am able to build and run the sample Run_video_slam.exe, but after run a few frames a debug assertion warning message popup. The warning message is attached with this issue.

reunanen, 2020-01-11:
How did you build? Using [this](https://github.com/reunanen/openvslam-build-vs)?

michealChin, 2020-01-12:
Yup, I follow the instruction exactly. Thanks for your great effort to convert it into vs project!

reunanen, 2020-01-12:
Oh, actually I _can_ reproduce this! I thought I couldn't but previously I just didn't wait long enough.

reunanen, 2020-01-12:
Should be fixed now. Just pull the latest [openvslam-build-vs](https://github.com/reunanen/openvslam-build-vs) and update the submodule (`git submodule update`).

reunanen, 2020-01-12:
(Or alternatively, cherry-pick [c04bfa09](https://github.com/reunanen/openvslam/commit/c04bfa09de5efe3f099ee31aa166629bd70e81db).)

reunanen, 2020-02-27:
@shinsumicco As suggested by [the bug report](https://github.com/xdspacelab/openvslam/issues/221#issue-547862411), this problem can be reproduced by running a Debug build in Visual Studio. (But as I [wrote already](
https://github.com/xdspacelab/openvslam/pull/217#discussion_r365566656), I believe this bug is more general, i.e. not really specific to Visual Studio or Windows.)

PR #258 (2020-02-28): Use correct end iterator in case keyframe is added
Resolves #221.

(Creating a separate PR, as per @shinsumicco's comment [here](https://github.com/xdspacelab/openvslam/pull/217#discussion_r385064255).)

The problem appears to be that after adding a local keyframe, the previously acquired `end` iterator isn't valid anymore ([if the vector needs to re-allocate](https://bytes.com/topic/c/answers/584854-does-push_back-invalidate-existing-vector-iterators)).

Note that when it's done like this, it means that the newly added keyframes _will_ be processed during this loop. Not sure if that is the in [... cut]
closed 2020-02-28 by shinsumicco