xdspacelab/openvslam #227: Error when saving trajectory by using --eval-log

anakin0620, 2020-01-18:
[C] cannot create a file at frame_trajectory.txt
terminate called after throwing an instance of 'std::runtime_error'
  what():  cannot create a file at frame_trajectory.txt
*** Aborted at 1579319730 (unix time) try "date -d @1579319730" if you are using GNU date ***
PC: @     0x7f4405f8de97 gsignal
*** SIGABRT (@0x3e800002f5f) received by PID 12127 (TID 0x7f44095d8bc0) from PID 12127; stack trace: ***
    @     0x7f4408925890 (unknown)
    @     0x7f4405f8de97 gsignal
    @     0x7f4405f8f801 abort
    @     0x7f44065e4957 (unknown)
    @     0x7f44065eaab6 (unknown)
    @     0x7f44065eaaf1 std::terminate()
    @     0x7f44065ead24 __cxa_throw
    @     0x7f4408c4b114 openvslam::io::trajectory_io::save_frame_trajectory()
    @     0x7f4408bd0e15 openvslam::system::save_frame_trajectory()
    @     0x5588e558b672 (unknown)
    @     0x5588e558893a (unknown)
    @     0x7f4405f70b97 __libc_start_main
    @     0x5588e558912a (unknown)
Aborted (core dumped)

Thank you in advance, any suggestion is appreciated!

anakin0620, 2020-01-18:
[image]

anakin0620, 2020-01-18:
I know this is another stupid question I asked. It's been solved by adding permission to add/delete file. I'll keep this here just in case someone else waste 1 hour googling like me. It's fixed now.

closed 2020-01-18 by anakin0620

litingpan, 2020-07-13:
Hello!Could you tell me how to set to save the trajectory?I use this command:
./build/run_video_slam \
    -v orb_vocab/orb_vocab.dbow2 \
    -c /mnt/hgfs/ubuntushare/data/equirectangular/aist_living_lab_1/config.yaml \
    -m /mnt/hgfs/ubuntushare/data/equirectangular/aist_living_lab_1/video.mp4 \
    --frame-skip 3 \
    --eval-log \
    --map-db aist_living_lab_1_map.msg
but can't save trajectory。