xdspacelab/openvslam #231: ROS package installation error

jungladicitta, 2020-01-22:
When I do catkin_make I see error:
```
Base path: /home/r2software/dev/openvslam/ros
Source space: /home/r2software/dev/openvslam/ros/src
Build space: /home/r2software/dev/openvslam/ros/build
Devel space: /home/r2software/dev/openvslam/ros/devel
Install space: /home/r2software/dev/openvslam/ros/install
####
#### Running command: "cmake /home/r2software/dev/openvslam/ros/src -DBUILD_WITH_MARCH_NATIVE=ON -DUSE_PANGOLIN_VIEWER=ON -DUSE_SOCKET_PUBLISHER=OFF -DUSE_STACK_TRACE_LOGGER=ON -DBOW_FRAMEWORK=DBoW2 -DCATKIN_DEVEL_PREFIX=/home/r2software/dev/openvslam/ros/devel -DCMAKE_INSTALL_PREFIX=/home/r2software/dev/openvslam/ros/install -G Unix Makefiles" in "/home/r2software/dev/openvslam/ros/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/r2software/dev/openvslam/ros/devel
-- Using CMAKE_PREFIX_PATH: /home/r2software/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/r2software/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/r2software/dev/openvslam/ros/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.18
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - cv_bridge
-- ~~  - openvslam
-- ~~  - publisher
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'cv_bridge'
-- ==> add_subdirectory(cv_bridge)
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found version "2.7.12") 
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   python
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable version "2.7.12", minimum required is "2.7") 
-- +++ processing catkin package: 'openvslam'
-- ==> add_subdirectory(openvslam)
-- Root directory of OpenVSLAM: /home/r2software/dev/openvslam
-- Source directory of OpenVSLAM: /home/r2software/dev/openvslam/src
-- Library directory of OpenVSLAM: /home/r2software/dev/openvslam/build/lib
-- Build type: Release
-- Address/Memory sanitizer: DISABLED
-- Stack trace logger: ENABLED
-- No preference for use of exported glog CMake configuration set, and no hints for include/library directories provided. Defaulting to preferring an installed/exported glog CMake configuration if available.
-- Failed to find installed glog CMake configuration, searching for glog build directories exported with CMake.
-- Failed to find an installed/exported CMake configuration for glog, will perform search for installed glog components.
-- Google Perftools: DISABLED
-- Architecture-aware optimization (-march=native): ENABLED
-- C++11 support: OK (-std=c++17)
-- Use OpenCV 3.3.1
-- BoW framework: DBoW2 (found in /usr/local/include)
-- Viewer for examples: pangolin viewer
-- +++ processing catkin package: 'publisher'
-- ==> add_subdirectory(publisher)
-- Root directory of OpenVSLAM: /home/r2software/dev/openvslam
-- Build type: Release
-- Address/Memory sanitizer: DISABLED
-- Architecture-aware optimization (-march=native): ENABLED
-- C++11 support: OK (-std=c++17)
-- Use OpenCV 3.3.1
-- Configuring done
-- Generating done
-- Build files have been written to: /home/r2software/dev/openvslam/ros/build
####
#### Running command: "make -j8 -l8" in "/home/r2software/dev/openvslam/ros/build"
####
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[ 20%] Built target cv_bridge
[ 33%] Built target video_publisher
[ 53%] Built target cv_bridge_boost
[ 73%] Built target image_publisher
[ 86%] Linking CXX executable /home/r2software/dev/openvslam/ros/devel/lib/openvslam/run_localization
[ 86%] Linking CXX executable /home/r2software/dev/openvslam/ros/devel/lib/openvslam/run_slam
/home/r2software/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `omp_destroy_lock'
/home/r2software/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `g2o::OptimizableGraph::addVertex(g2o::OptimizableGraph::Vertex*, g2o::HyperGraph::Data*)'
/home/r2software/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `omp_init_lock'
/home/r2software/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `omp_unset_lock'
/home/r2software/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `omp_set_lock'
/home/r2software/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(std::unique_ptr<g2o::Solver, std::default_delete<g2o::Solver> >)'
/home/r2software/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `g2o::OptimizableGraph::addEdge(g2o::OptimizableGraph::Edge*)'
collect2: error: ld returned 1 exit status
openvslam/src/CMakeFiles/run_localization.dir/build.make:152: recipe for target '/home/r2software/dev/openvslam/ros/devel/lib/openvslam/run_localization' failed
/make[2]: *** [/home/r2software/dev/openvslam/ros/devel/lib/openvslam/run_localization] Error 1
home/r2software/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `omp_destroy_lock'
/home/r2software/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `g2o::OptimizableGraph::addVertex(g2o::OptimizableGraph::Vertex*, g2o::HyperGraph::Data*)'
/home/r2software/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `omp_init_lock'
/home/r2software/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `omp_unset_lock'
/home/r2software/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `omp_set_lock'
/home/r2software/dev/openvslam/build/lib/libopenvslam.so: undefinedCMakeFiles/Makefile2:1989: recipe for target 'openvslam/src/CMakeFiles/run_localization.dir/all' failed
 reference make[1]: *** [openvslam/src/CMakeFiles/run_localization.dir/all] Error 2
to `g2o::OptimizationAlgorithmLevenberg::make[1]: *** Waiting for unfinished jobs....
OptimizationAlgorithmLevenberg(std::unique_ptr<g2o::Solver, std::default_delete<g2o::Solver> >)'
/home/r2software/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `g2o::OptimizableGraph::addEdge(g2o::OptimizableGraph::Edge*)'
collect2: error: ld returned 1 exit status
openvslam/src/CMakeFiles/run_slam.dir/build.make:152: recipe for target '/home/r2software/dev/openvslam/ros/devel/lib/openvslam/run_slam' failed
make[2]: *** [/home/r2software/dev/openvslam/ros/devel/lib/openvslam/run_slam] Error 1
CMakeFiles/Makefile2:2051: recipe for target 'openvslam/src/CMakeFiles/run_slam.dir/all' failed
make[1]: *** [openvslam/src/CMakeFiles/run_slam.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

I found this [issue](https://github.com/xdspacelab/openvslam/issues/209) and I supposed that my issue is related to g2o.
So, I tried to reinstall g2o, to do that I cloned g2o project again and did all the steps except for `git checkout`.
Then I tried `catkin_make` and I've got what you see above.
What can I do to do catkin_make successfully? Have I reinstalled g2o?

shinsumicco, 2020-01-22:
@jungladicitta z
It seems `ld` failed to link OpenMP to the library file.
Could you reconfigure and rebuild OpenVSLAM / ROS bindings with `-DUSE_OPENMP=OFF` option ?

jungladicitta, 2020-01-23:
@shinsumicco I tried 
```
cmake \
    -DBUILD_WITH_MARCH_NATIVE=ON \
    -DUSE_PANGOLIN_VIEWER=ON \
    -DUSE_SOCKET_PUBLISHER=OFF \
    -DUSE_STACK_TRACE_LOGGER=ON \
    -DBOW_FRAMEWORK=DBoW2 \
    -DBUILD_TESTS=ON \
    -DUSE_OPENMP=OFF \
    ..
make -j4
```
in openvslam/build directory, and then
```
catkin_make \
    -DBUILD_WITH_MARCH_NATIVE=ON \
    -DUSE_PANGOLIN_VIEWER=ON \
    -DUSE_SOCKET_PUBLISHER=OFF \
    -DUSE_STACK_TRACE_LOGGER=ON \
    -DBOW_FRAMEWORK=DBoW2 \
    -DUSE_OPENMP=OFF
```
in openvslam/ros directory.
Nothing has actually changed. I still see similar errors:
```
Base path: /home/yuki/dev/openvslam/ros
Source space: /home/yuki/dev/openvslam/ros/src
Build space: /home/yuki/dev/openvslam/ros/build
Devel space: /home/yuki/dev/openvslam/ros/devel
Install space: /home/yuki/dev/openvslam/ros/install
####
#### Running command: "make cmake_check_build_system" in "/home/yuki/dev/openvslam/ros/build"
####
####
#### Running command: "make -j8 -l8" in "/home/yuki/dev/openvslam/ros/build"
####
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target roscpp_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[ 20%] Built target cv_bridge
[ 40%] Built target cv_bridge_boost
[ 60%] Built target image_publisher
[ 73%] Built target video_publisher
[ 80%] Linking CXX executable /home/yuki/dev/openvslam/ros/devel/lib/openvslam/run_slam
[ 86%] Linking CXX executable /home/yuki/dev/openvslam/ros/devel/lib/openvslam/run_localization
/home/yuki/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `omp_destroy_lock'
/home/yuki/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `g2o::OptimizableGraph::addVertex(g2o::OptimizableGraph::Vertex*, g2o::HyperGraph::Data*)'
/home/yuki/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `omp_init_lock'
/home/yuki/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `omp_unset_lock'
/home/yuki/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `omp_set_lock'
/home/yuki/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(std::unique_ptr<g2o::Solver, std::default_delete<g2o::Solver> >)'
/home/yuki/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `g2o::OptimizableGraph::addEdge(g2o::OptimizableGraph::Edge*)'
/home/yuki/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `omp_destroy_lock'
/home/yuki/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `g2o::OptimizableGraph::addVertex(g2o::OptimizableGraph::Vertex*, g2o::HyperGraph::Data*)'
/home/yuki/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `omp_init_lock'
/home/yuki/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `omp_unset_lock'
/home/yuki/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `omp_set_lock'
/home/yuki/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(std::unique_ptr<g2o::Solver, std::default_delete<g2o::Solver> >)'
/home/yuki/dev/openvslam/build/lib/libopenvslam.so: undefined reference to `g2o::OptimizableGraph::addEdge(g2o::OptimizableGraph::Edge*)'
collect2: error: ld returned 1 exit status
collect2: error: ld returned 1 exit status
openvslam/src/CMakeFiles/run_slam.dir/build.make:152: recipe for target '/home/yuki/dev/openvslam/ros/devel/lib/openvslam/run_slam' failed
make[2]: *** [/home/yuki/dev/openvslam/ros/devel/lib/openvslam/run_slam] Error 1
openvslam/src/CMakeFiles/run_localization.dir/build.make:152: recipe for target '/home/yuki/dev/openvslam/ros/devel/lib/openvslam/run_localization' failed
make[2]: *** [/home/yuki/dev/openvslam/ros/devel/lib/openvslam/run_localization] Error 1
CMakeFiles/Makefile2:2051: recipe for target 'openvslam/src/CMakeFiles/run_slam.dir/all' failed
make[1]: *** [openvslam/src/CMakeFiles/run_slam.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1989: recipe for target 'openvslam/src/CMakeFiles/run_localization.dir/all' failed
make[1]: *** [openvslam/src/CMakeFiles/run_localization.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```
Maybe I am doing something wrong? What does mean reconfigure?

jungladicitta, 2020-01-24:
@shinsumicco I also tried to install openmp `sudo apt-get install libomp-dev`.
nothing has changed

shinsumicco, 2020-01-25:
Hmm...
Could you tell me about the information of your environment?

jungladicitta, 2020-01-27:
@shinsumicco 
hp probook 450 g6, Intel Core i7-8565U 
`sudo lshw -C display` gives me:
```
*-display               
       description: VGA compatible controller
       product: Intel Corporation
       vendor: Intel Corporation
       physical id: 2
       bus info: pci@0000:00:02.0
       version: 02
       width: 64 bits
       clock: 33MHz
       capabilities: pciexpress msi pm vga_controller bus_master cap_list rom
       configuration: driver=i915 latency=0
       resources: irq:129 memory:f0000000-f0ffffff memory:e0000000-efffffff ioport:4000(size=64) memory:c0000-dffff
```

ubuntu 16.04,
`uname -r`:
4.15.0-74-generic

ROS Kinetic

shinsumicco, 2020-01-27:
@jungladicitta 
Thanks for providing me the information.
I'm not sure about the cause of the failure, but I suggest something worth to try:

1. Check whether you can run SLAM without ROS bindings. If it is executable, it seems that CMake configuration for ROS bindings needs to be fixed.
1. Rebuild and reinstall g2o without OpenMP dependencies. According to the build log, it seems failed to link some g2o functions.

jungladicitta, 2020-01-28:
@shinsumicco Thanks!

1. SLAM without ROS is working ok.
2. I will try soon, and let you know.

jungladicitta, 2020-01-28:
@shinsumicco nope, still the same errors.
I went to my g2o/build directory, did there:
```
cmake \
    -DCMAKE_BUILD_TYPE=Release \
    -DCMAKE_INSTALL_PREFIX=/usr/local \
    -DCMAKE_CXX_FLAGS=-std=c++11 \
    -DBUILD_SHARED_LIBS=ON \
    -DBUILD_UNITTESTS=OFF \
    -DBUILD_WITH_MARCH_NATIVE=ON \
    -DG2O_USE_CHOLMOD=OFF \
    -DG2O_USE_CSPARSE=ON \
    -DG2O_USE_OPENGL=OFF \
    -DG2O_USE_OPENMP=OFF \
    ..
make -j4
sudo make install
```
then, I went to openvslam/ros directory and did there
```
catkin_make \
    -DBUILD_WITH_MARCH_NATIVE=ON \
    -DUSE_PANGOLIN_VIEWER=ON \
    -DUSE_SOCKET_PUBLISHER=OFF \
    -DUSE_STACK_TRACE_LOGGER=ON \
    -DBOW_FRAMEWORK=DBoW2
    -DUSE_OPENMP=OFF
```
Am I doing everything correct? Have g2o actually reinstalled? I am not good at Linux

timotheos, 2020-01-28:
Hi, I have the same errors, specifically:
```
undefined reference to `g2o::OptimizableGraph::addVertex(g2o::OptimizableGraph::Vertex*, g2o::HyperGraph::Data*)'
undefined reference to `g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(g2o::Solver*)'
```
On my system, it conflicted with a ros installed libg2o package. So I temporarily removed it with
``` 
sudo apt remove ros-kinetic-libg2o 
```
then ``` catkin_make ``` on my catkin workspace, and reinstalling install it again 
``` 
sudo apt install ros-kinetic-libg2o 
```
I'm not sure if it will work for you, but worth a shot!

jungladicitta, 2020-01-29:
@timotheos its a miracle! I am really appreciated for your help!
I removed `ros-kinetic-libg2o` as you said and then I tried
```
catkin_make \
    -DBUILD_WITH_MARCH_NATIVE=ON \
    -DUSE_PANGOLIN_VIEWER=ON \
    -DUSE_SOCKET_PUBLISHER=OFF \
    -DUSE_STACK_TRACE_LOGGER=ON \
    -DBOW_FRAMEWORK=DBoW2
    -DUSE_OPENMP=OFF
```
and there were no errors about g2o, but there still were errors about omp.
Then I removed `ros-kinetic-opencv3` and was asked to remove this list of ros-kinetic packages:
```
ros-kinetic-camera-calibration ros-kinetic-compressed-depth-image-transport ros-kinetic-compressed-image-transport ros-kinetic-costmap-converter ros-kinetic-cv-bridge ros-kinetic-depth-image-proc ros-kinetic-desktop ros-kinetic-desktop-full ros-kinetic-find-object-2d ros-kinetic-gazebo-plugins ros-kinetic-gazebo-ros-pkgs ros-kinetic-hector-compressed-map-transport ros-kinetic-image-geometry ros-kinetic-image-pipeline ros-kinetic-image-proc ros-kinetic-image-publisher ros-kinetic-image-rotate ros-kinetic-image-transport-plugins ros-kinetic-image-view ros-kinetic-opencv3 ros-kinetic-perception ros-kinetic-rgbd-launch ros-kinetic-rqt-common-plugins ros-kinetic-rqt-image-view ros-kinetic-simulators ros-kinetic-stereo-image-proc ros-kinetic-theora-image-transport ros-kinetic-vision-opencv ros-kinetic-viz
```
I said yes to remove them and then tried `catkin_make` again and I got a lot of errors. Then I installed `ros-kinetic-opencv3` back and then catkin_make with all that flags again and it worked!

-------------------------

Then I tried to run openvslam via ROS on nu_eng2_corridor_1 Sample Dataset. It successfully ran but it become freezed after one image frame. In the terminal I see:
```
jessicaalba@hp-probook:~$ rosrun openvslam run_slam -v /home/jessicaalba/Downloads/orb_vocab/orb_vocab.dbow2 -c /home/jessicaalba/Downloads/nu_eng2_corridor_1/config.yaml 
[2020-01-29 11:37:33.732] [I] config file loaded: /home/jessicaalba/Downloads/nu_eng2_corridor_1/config.yaml
  ___               __   _____ _      _   __  __ 
 / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
| (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
 \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
      |_|                                        

Copyright (C) 2019,
National Institute of Advanced Industrial Science and Technology (AIST)
All rights reserved.

This is free software,
and you are welcome to redistribute it under certain conditions.
See the LICENSE file.

Camera Configuration:
- name: LUMIX DMC-GX8 8mm fisheye 1K
- setup: Monocular
- fps: 30
- cols: 960
- rows: 540
- color: RGB
- model: Fisheye
  - fx: 441.73
  - fy: 442.521
  - cx: 480.357
  - cy: 275.49
  - k1: -0.030687
  - k2: -0.00334345
  - k3: -0.00288174
  - k4: -0.000591742
  - min x: -1688.78
  - max x: 2615.94
  - min y: -968.534
  - max y: 1342.99
ORB Configuration:
- number of keypoints: 2000
- scale factor: 1.2
- number of levels: 8
- initial fast threshold: 20
- minimum fast threshold: 7

[2020-01-29 11:37:33.733] [I] loading ORB vocabulary: /home/jessicaalba/Downloads/orb_vocab/orb_vocab.dbow2
[2020-01-29 11:37:33.995] [I] startup SLAM system
[2020-01-29 11:37:33.996] [I] start mapping module
[2020-01-29 11:37:33.996] [I] start global optimization module

** (run_slam:12126): WARNING **: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-HTacLQhGdY: Connection refused
[2020-01-29 11:37:34.399] [I] initialization succeeded with F
```
I tried to install that list of packaged that I removed with opencv3 and it didnt helped.
I think its another issue.

Have you confronted with similar error?

closed 2020-01-29 by jungladicitta

shinsumicco, 2020-01-29:
@timotheos 
Thank you for providing us the useful information!
@jungladicitta 
I'm glad to hear that!

dimaxano, 2020-04-08:
Hi!

For me problem with with `undefined reference to `g2o::...` is quite annoying 'cause I am making some changes in ROS version of Open VSLAM that is why I need to reinstall remove\install libg2o from ros directory
I've solved this just by installing g2o to `/opt/ros/ros_version` dir (by setting cmake param `-DCMAKE_INSTALL_PREFIX`). Not sure if that won't broke some other package, but for now it works :)