xdspacelab/openvslam #233: How can I create a model for a new panoramic camera?

mans-nctech, 2020-01-28:
I like to create a model for iSTAR Pulsar https://www.nctechimaging.com/capture/hardware/

Where should I start? 
Is there any documentation on the parameters in model file and how can I find them for a camera?

shinsumicco, 2020-01-29:
@mans-nctech 
Thanks for the interest in this project.
For panoramic (equirectangular) images, you should configure image size, FPS, and mask areas according to the camera you'll use.
Please see the sample configuration files posted on [here](https://openvslam.readthedocs.io/en/master/simple_tutorial.html#equirectangular-datasets).

mans-nctech, 2020-01-29:
what is mask areas and how can I calculate it for a camera?