xdspacelab/openvslam #234: run_slam become freezed after one image frame: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-HTacLQhGdY: Connection refused

jungladicitta, 2020-01-29:
@timotheos its a miracle! I am really appreciated for your help!
I removed `ros-kinetic-libg2o` as you said and then I tried
```
catkin_make \
    -DBUILD_WITH_MARCH_NATIVE=ON \
    -DUSE_PANGOLIN_VIEWER=ON \
    -DUSE_SOCKET_PUBLISHER=OFF \
    -DUSE_STACK_TRACE_LOGGER=ON \
    -DBOW_FRAMEWORK=DBoW2
    -DUSE_OPENMP=OFF
```
and there were no errors about g2o, but there still were errors about omp.
Then I removed `ros-kinetic-opencv3` and was asked to remove this list of ros-kinetic packages:
```
ros-kinetic-camera-calibration ros-kinetic-compressed-depth-image-transport ros-kinetic-compressed-image-transport ros-kinetic-costmap-converter ros-kinetic-cv-bridge ros-kinetic-depth-image-proc ros-kinetic-desktop ros-kinetic-desktop-full ros-kinetic-find-object-2d ros-kinetic-gazebo-plugins ros-kinetic-gazebo-ros-pkgs ros-kinetic-hector-compressed-map-transport ros-kinetic-image-geometry ros-kinetic-image-pipeline ros-kinetic-image-proc ros-kinetic-image-publisher ros-kinetic-image-rotate ros-kinetic-image-transport-plugins ros-kinetic-image-view ros-kinetic-opencv3 ros-kinetic-perception ros-kinetic-rgbd-launch ros-kinetic-rqt-common-plugins ros-kinetic-rqt-image-view ros-kinetic-simulators ros-kinetic-stereo-image-proc ros-kinetic-theora-image-transport ros-kinetic-vision-opencv ros-kinetic-viz
```
I said yes to remove them and then tried `catkin_make` again and I got a lot of errors. Then I installed `ros-kinetic-opencv3` back and then catkin_make with all that flags again and it worked!

-------------------------

Then I tried to run openvslam via ROS on nu_eng2_corridor_1 Sample Dataset. It successfully ran but it become freezed after one image frame. In the terminal I see:
```
jessicaalba@hp-probook:~$ rosrun openvslam run_slam -v /home/jessicaalba/Downloads/orb_vocab/orb_vocab.dbow2 -c /home/jessicaalba/Downloads/nu_eng2_corridor_1/config.yaml 
[2020-01-29 11:37:33.732] [I] config file loaded: /home/jessicaalba/Downloads/nu_eng2_corridor_1/config.yaml
  ___               __   _____ _      _   __  __ 
 / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
| (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
 \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
      |_|                                        

Copyright (C) 2019,
National Institute of Advanced Industrial Science and Technology (AIST)
All rights reserved.

This is free software,
and you are welcome to redistribute it under certain conditions.
See the LICENSE file.

Camera Configuration:
- name: LUMIX DMC-GX8 8mm fisheye 1K
- setup: Monocular
- fps: 30
- cols: 960
- rows: 540
- color: RGB
- model: Fisheye
  - fx: 441.73
  - fy: 442.521
  - cx: 480.357
  - cy: 275.49
  - k1: -0.030687
  - k2: -0.00334345
  - k3: -0.00288174
  - k4: -0.000591742
  - min x: -1688.78
  - max x: 2615.94
  - min y: -968.534
  - max y: 1342.99
ORB Configuration:
- number of keypoints: 2000
- scale factor: 1.2
- number of levels: 8
- initial fast threshold: 20
- minimum fast threshold: 7

[2020-01-29 11:37:33.733] [I] loading ORB vocabulary: /home/jessicaalba/Downloads/orb_vocab/orb_vocab.dbow2
[2020-01-29 11:37:33.995] [I] startup SLAM system
[2020-01-29 11:37:33.996] [I] start mapping module
[2020-01-29 11:37:33.996] [I] start global optimization module

** (run_slam:12126): WARNING **: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-HTacLQhGdY: Connection refused
[2020-01-29 11:37:34.399] [I] initialization succeeded with F
```
I tried to install that list of packaged that I removed with opencv3 and it didnt helped.
I think its another issue.

Have you confronted with similar error?

_Originally posted by @jungladicitta in https://github.com/xdspacelab/openvslam/issues/231#issuecomment-579591801_

shinsumicco, 2020-01-29:
@jungladicitta 
I haven't ever encountered such a error.

What kind of viewer do you try to use?
If you do with PangolinVIewer, does the GUI display launch?

jungladicitta, 2020-01-29:
@shinsumicco pangolin, GUI display does launch. becomes freezed in less than a second

timotheos, 2020-01-29:
@jungladicitta @shinsumicco 
I have encountered this error before, but I can't quite remember how I handled it. I fixed it by either  configuring the camera.yaml file incorrectly or recompiling it with ``` BUILD_WITH_MARCH_NATIVE=OFF ```

I'll get back to you if I encounter the issue again :+1: