xdspacelab/openvslam #237: Compiling examples separately ? (How to include openvslam)

alexs7, 2020-01-30:
I have successfully build/compiled openvslam. Now I am trying to compile an example file such as _run_image_localization.cc_ from [here](https://github.com/xdspacelab/openvslam/tree/master/example), by simple using `g++ run_image_localization.cc`

I get this:

```
run_image_localization.cc:9:10: fatal error: 'openvslam/system.h' file not found
#include "openvslam/system.h"
```

How do I link the library so it compiles?

klintan, 2020-02-09:
The examples are built by default if you just follow the instructions on the documentation-site.
`CMakeLists.txt` file:

```
set(BUILD_EXAMPLES ON CACHE BOOL "Build examples")
if(BUILD_EXAMPLES)
    add_subdirectory(example)
endif()
```
However if the question is how to add openvslam as a separate library, you could take a look at the ROS-folder `CMakeLists.txt` file found here
https://github.com/xdspacelab/openvslam/blob/master/ros/src/openvslam/CMakeLists.txt

alexs7, 2020-02-09:
@klintan Yes I got that, but I wanted to compile a single file that includes openvslam. I think that link might help. Thanks!