xdspacelab/openvslam #239: Compile errors on ubuntu 18 and macOS Mojave

monschine, 2020-02-03:
HI,
congrats for the framework !
While I succeeded compiling the software in ubuntu 16, it was not possible in ubuntu 18 or macOS Mojave.
The issue is from a g2o make error. Anyone experienced that and found how to solve it ?

Thanks in advance and sorry if that's a nob question.

klintan, 2020-02-08:
Please post the output from your compilation failure, otherwise it's very difficult to triage and troubleshoot the issue.

monschine, 2020-02-09:
Hi @klintan, just following the building instructions I get this error :

Scanning dependencies of target core
[ 11%] Building CXX object g2o/core/CMakeFiles/core.dir/hyper_graph.cpp.o
[ 11%] Building CXX object g2o/core/CMakeFiles/core.dir/matrix_structure.cpp.o
[ 13%] Building CXX object g2o/core/CMakeFiles/core.dir/hyper_graph_action.cpp.o
[ 13%] Building CXX object g2o/core/CMakeFiles/core.dir/marginal_covariance_cholesky.cpp.o
In file included from /Users/mosnchine/Documents/openvslam/g2o/g2o/core/hyper_graph_action.cpp:28:
In file included from /Users/mosnchine/Documents/openvslam/g2o/g2o/core/optimizable_graph.h:34:
/Users/mosnchine/Documents/openvslam/g2o/g2o/core/openmp_mutex.h:33:10: fatal error: 
      'omp.h' file not found
#include <omp.h>
         ^~~~~~~
1 error generated.
make[2]: *** [g2o/core/CMakeFiles/core.dir/hyper_graph_action.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [g2o/core/CMakeFiles/core.dir/all] Error 2
make: *** [all] Error 2

Thanks for taking a look

klintan, 2020-02-09:
Have you tried running
`brew install libomp` 

If that doesn't work it seems the `clang`-compiler could be the issue. You might need to switch to gcc. But I would try the `brew install libomp` first :) 

Reference: 
https://stackoverflow.com/questions/25990296/how-to-include-omp-h-in-os-x

monschine, 2020-02-09:
Thanks, 
I already installed libomp, via brew but it still gets the same error.

klintan, 2020-02-09:
I haven't had any issues actually, strange. 

What versions of Clang, gcc and g++ are you using ? 

Have you tried the other suggestion; `cmake -DCMAKE_C_COMPILER=/usr/bin/gcc -DCMAKE_CXX_COMPILER=/usr/bin/g++` 
Forcing it to not use clang.

monschine, 2020-02-09:
Thanks,
that made it work
-DCMAKE_C_COMPILER=/usr/local/bin/gcc-9 \
-DCMAKE_CXX_COMPILER=/usr/local/bin/g++-9 \

closed 2020-02-09 by monschine

RashidLadj, 2020-06-24:
OS X native compiler does not support OPENMP 

Using Clang in LLVM would fix this error. Run cmake like the following:
```cmake .. -DCMAKE_C_COMPILER=/usr/local/opt/llvm/bin/clang -DCMAKE_CXX_COMPILER=/usr/local/opt/llvm/bin/clang++
```
so for compile g2o for OpenVSlam , you would use this script:

```cd /path/to/working/dir
git clone https://github.com/RainerKuemmerle/g2o.git
cd g2o
git checkout 9b41a4ea5ade8e1250b9c1b279f3a9c098811b5a
mkdir build && cd build
cmake \
    -DCMAKE_BUILD_TYPE=Release \
    -DCMAKE_INSTALL_PREFIX=/usr/local \
    -DCMAKE_CXX_FLAGS=-std=c++11 \
    -DBUILD_SHARED_LIBS=ON \
    -DBUILD_UNITTESTS=OFF \
    -DBUILD_WITH_MARCH_NATIVE=ON \
    -DG2O_USE_CHOLMOD=OFF \
    -DG2O_USE_CSPARSE=ON \
    -DG2O_USE_OPENGL=OFF \
    -DG2O_USE_OPENMP=ON \
    .. -DCMAKE_C_COMPILER=/usr/local/opt/llvm/bin/clang -DCMAKE_CXX_COMPILER=/usr/local/opt/llvm/bin/clang++
make -j4
make install
```

RashidLadj, 2020-06-24:
The native OS X compiler does not support OPENMP 

Using Clang in LLVM would correct this error. Execute cmake by adding at the end:

```
-DCMAKE_C_COMPILER=/usr/local/opt/llvm/bin/clang -DCMAKE_CXX_COMPILER=/usr/local/opt/llvm/bin/clang++
```
So to compile g2o for OpenVSlam, you need to use this script:

```
cd /path/to/working/dir
git clone https://github.com/RainerKuemmerle/g2o.git
cd g2o
git checkout 9b41a4ea5ade8e1250b9c1b279f3a9c098811b5a
mkdir build && cd build
cmake \
    -DCMAKE_BUILD_TYPE=Release \
    -DCMAKE_INSTALL_PREFIX=/usr/local \
    -DCMAKE_CXX_FLAGS=-std=c++11 \
    -DBUILD_SHARED_LIBS=ON \
    -DBUILD_UNITTESTS=OFF \
    -DBUILD_WITH_MARCH_NATIVE=ON \
    -DG2O_USE_CHOLMOD=OFF \
    -DG2O_USE_CSPARSE=ON \
    -DG2O_USE_OPENGL=OFF \
    -DG2O_USE_OPENMP=ON \
    .. -DCMAKE_C_COMPILER=/usr/local/opt/llvm/bin/clang -DCMAKE_CXX_COMPILER=/usr/local/opt/llvm/bin/clang++
make -j4
make install
```

elfelround, 2020-08-12:
> _OpenVSLAM requires a C++11-compliant compiler. It relies on several open-source libraries as shown below._

im a bit of a noob with gcc and c++ on ubuntu, do i need to explicitly install any of those according to this line?