xdspacelab/openvslam #240: Run in real time on Android device

monschine, 2020-02-03:
Hi, 

Is it possible to build an android app, and ./run_video_slam in real time ?
If it's possible, does the device would need a high specs to achieve that goal ?
Any hint on how to do that ?

TheBricktop, 2020-02-15:
This is valid question, I would like to know that too.

BijayanBhattarai, 2020-03-13:
i am waiting for answer

williamliuok, 2020-03-20:
We had a really hard time to make it runnable on android, but we found it quite slow, We are still working on fugure out what's went worng

TheBricktop, 2020-03-22:
@williamliuok  can You share Your work? Im very interested in making this work on more devices.

williamliuok, 2020-03-25:
@TheBricktop Yeah,I can share my project on Github later, you guys might need it.

iegorval, 2020-09-12:
@williamliuok In the end, did you have any luck running it on Android at some (near) real-time speed?

jeronimoagullo, 2021-01-21:
@williamliuok Are there news about running it on Android?