xdspacelab/openvslam #244: What is the format of frame_trajectory.txt ?

IaroslavS, 2020-02-10:
Hi !
There are 8 numbers in file frame_trajectory.txt. What do they mean ? As far as I know an SE3 matrix consists of 16 number, and 12 of them are significant, because the last row of the matrix consists of 0,0,0,1.

Shashika007, 2020-02-11:
@IaroslavS frame_trajectory.txt contains timestamp, 3dof pose, orientation as Quaternion. you can compute SE3 matrix from the 3dof pose (x,y,z) and orientation-->SO3.

IaroslavS, 2020-02-11:
> @IaroslavS frame_trajectory.txt contains timestamp, 3dof pose, orientation as Quaternion. you can compute SE3 matrix from the 3dof pose (x,y,z) and orientation-->SO3.

Thanks.
There's a mistake in repository in file run_kitti_slam.cc - there should be "KITTI" instead of "TUM"

squaresoft2015, 2020-02-22:
I use the evo tools to evaluate the result. However, it promots [ERROR] KITTI pose files must have 12 entries per row and no trailing delimiter at the end of the rows (space). So  how can I make the output as the format required above?

shinsumicco, 2020-02-22:
Trajectory files will be written as the TUM-style trajectory format.
So when using `evo` please specify a subcommand  like this:
```sh
$ evo_traj tum keyframe_trajectory.txt -p
```

shinsumicco, 2020-02-22:
@IaroslavS You can browse the trajectory format [here](https://vision.in.tum.de/data/datasets/rgbd-dataset/file_formats).

squaresoft2015, 2020-02-22:
I specified the tum parameter in evo, an error message response:
[ERROR] TUM trajectory files must have 8 entries per row and no trailing delimiter at the end of the rows (space)

shinsumicco, 2020-02-22:
@squaresoft2015 
I could not reproduce it on my environment.
Could you show me some beginning lines of the trajectory file?

squaresoft2015, 2020-02-22:
The trajectory was generated by openvslam in KITTI02 sequences. Here is the beginning of the trajectory file:
0 0 0 0 0 0 0 1
0.1036117 0.00281646292 -0.0211759378 -1.51789995 0.00148783212 -0.000672189489 -0.00123988552 0.999997899
0.2073529 0.0192833077 -0.032919165 -3.01159644 0.00189333686 -0.00140504685 -0.00129932359 0.999996376
0.3111179 0.0333557922 -0.0513078082 -4.53333794 0.00177650121 -0.00214249078 -0.000889985888 0.999995731
0.4147131 0.0498640708 -0.0667355578 -6.03943024 0.00169346325 -0.00296813085 -0.000533904514 0.999994019
0.5199468 0.0681946093 -0.0809641733 -7.54225432 0.00181506802 -0.00358438031 -0.000474826148 0.999991816
0.6220911 0.082230531 -0.0970788809 -9.03818554 0.0020657405 -0.00455828999 -0.000812271917 0.999987147
0.725703 0.105753093 -0.110677316 -10.5361619 0.00165387362 -0.00533598955 -0.00153263116 0.999983221

shinsumicco, 2020-02-22:
@squaresoft2015 Thanks.
`evo` on my environment can load the trajectory you'd posted.
```
$ evo_traj tum traj.txt --verbose
--------------------------------------------------------------------------------
Loaded 8 stamps and poses from: traj.txt
--------------------------------------------------------------------------------
name:	traj
infos:
	duration (s)	0.725703
	nr. of poses	8
	path length (m)	10.537372946784544
	pos_end (m)	[  0.10575309  -0.11067732 -10.5361619 ]
	pos_start (m)	[0. 0. 0.]
	t_end (s)	0.725703
	t_start (s)	0.0
```
Please try to uninstall and re-install `evo`.

squaresoft2015, 2020-02-22:
Yes, the trajectory can be loaded, but not plotted in image file. When I execute the command as you have done, the result is the same:
(evo) ubuntu@ubuntu-Z170-HD3:~/Downloads/plot$ evo_traj tum frame_trajectory.txt --verbose
--------------------------------------------------------------------------------
Loaded 2232 stamps and poses from: frame_trajectory.txt
--------------------------------------------------------------------------------
name:	frame_trajectory
infos:
	duration (s)	466.7824
	nr. of poses	2232
	path length (m)	4130.440459749678
	pos_end (m)	[-717.631093   -41.4160494  437.771655 ]
	pos_start (m)	[0. 0. 0.]
	t_end (s)	466.7824
	t_start (s)	0.0
But if I want to plot it with the ground truth with the following command:
(evo) ubuntu@ubuntu-Z170-HD3:~/Downloads/plot$ evo_traj tum frame_trajectory.txt --ref=02.txt -p --plot_mode=xz
[ERROR] TUM trajectory files must have 8 entries per row and no trailing delimiter at the end of the rows (space)
The error is the same.

shinsumicco, 2020-02-22:
@squaresoft2015
According to the error message, one or more lines of the trajectory file don't have eight entries.
Please confirm that all of the lines have 8 entries with the following shell script:
```
$ IFS=$'\n'; for LINE in $(cat traj.txt); do [ $(echo $LINE | wc -w) == 8 ] || echo "error: $LINE"; done
```
If the trajectory file is NOT in the TUM-style format, the following message will be appeared:
```
$ IFS=$'\n'; for LINE in $(cat traj.txt); do [ $(echo $LINE | wc -w) == 8 ] || echo "error: $LINE"; done
error: xxx yyy zzz www aaa bbb ccc
```

squaresoft2015, 2020-02-22:
I have executed the script and no error message generated.
(evo) ubuntu@ubuntu-Z170-HD3:~/Downloads/plot$ IFS=$'\n'; for LINE in $(cat frame_trajectory.txt); do [ $(echo $LINE | wc -w) == 8 ] || echo "error: $LINE"; done
(evo) ubuntu@ubuntu-Z170-HD3:~/Downloads/plot$

shinsumicco, 2020-02-22:
@squaresoft2015 Please remove the last row of the trajectory file if it is a blank line.

squaresoft2015, 2020-02-22:
> @squaresoft2015 Please remove the last row of the trajectory file if it is a blank line.

There is no blank line in the trajectory file or ground truth. The only difference is that the trajectory file has 8 digits per row and ground truth has 12.

shinsumicco, 2020-02-22:
@squaresoft2015 Please confirm that the ground-truth trajectory file is in the TUM-style.
In my knowledge, the official GT trajectory files of KITTI are NOT in the TUM-style, but the KITTI-style.

shinsumicco, 2020-02-22:
You can convert KITTI-style files to TUM style ones using `evo`.

squaresoft2015, 2020-02-22:
> You can convert KITTI-style files to TUM style ones using `evo`:
> 
> ```
> $ evo_traj kitti traj.txt --save_as_tum
> ```

Yes, I tried to save the KITTI ground truth as TUM format. But it still prompt error message:
(evo) ubuntu@ubuntu-Z170-HD3:~/Downloads/plot$ evo_traj kitti frame_trajectory.txt --save_as_tum
[ERROR] KITTI pose files must have 12 entries per row and no trailing delimiter at the end of the rows (space)
I have checked the ground truth file and it is 12 entries per row...

shinsumicco, 2020-02-22:
@squaresoft2015 Trajectory files created by OpenVSLAM are in the TUM-style.
So first you should convert `frame_trajectory.txt` to the KITTI-style, ant then compare it with the official GT using `evo` (`$ evo kitti --ref GT.txt converted_frame_trajectory.kitti -p`).

Or, you can do it alternatively by converting the GT files to TUM-style ones first, then comparing `frame_trajectory.txt` with the converted GT file.

squaresoft2015, 2020-02-22:
Thank you so much. I have converted the TUM-style to KITTI-style and compare it with the ground truth. But the curve has no matching point. I have uploaded the image to Baidu net-disk:
Link: https://pan.baidu.com/s/1ijinp3mQ_ngTvZePtItxIw 
Request code: jgko
It should not be like that.....

shinsumicco, 2020-02-22:
@squaresoft2015 
Please try to add `--align` and `--correct_scale` options for `evo`.
```
$ evo_traj tum -h
(snip)
algorithm options:
  -a, --align           alignment with Umeyama's method (no scale) - requires --ref
  -s, --correct_scale   scale correction with Umeyama's method - requires --ref
(snip)
```

squaresoft2015, 2020-02-22:
> @squaresoft2015
> Please try to add `--align` and `--correct_scale` options for `evo`.
>
> ```
> $ evo_traj tum -h
> (snip)
> algorithm options:
>   -a, --align           alignment with Umeyama's method (no [...]

Yes, I tried the --align and --correct_scale option and it prompt:
[ERROR] data matrices must have the same shape

shinsumicco, 2020-02-22:
@squaresoft2015 Please see [here](https://github.com/MichaelGrupp/evo/issues/152#issuecomment-473660536).
According to the Michael's reply, you should compare the GT and the estimated trajectories in the TUM-format, which has timestamps for each line.

squaresoft2015, 2020-02-22:
> @squaresoft2015 Please see [here](https://github.com/MichaelGrupp/evo/issues/152#issuecomment-473660536).
> According to the Michael's reply, you should compare the GT and the estimated trajectories in the TUM-format, which has timestamps for each line.

I have added the timestamp to KITTI ground truth and compared with frame_trajectory.txt generated by Openvslam. The result is as follows:
Link: https://pan.baidu.com/s/1PmHZWMOnGP2xLTCM6_PCEQ 
Request code: 21if

Is it seems correct?

shinsumicco, 2020-02-23:
It seems correct except for jumping of the trajectory.

squaresoft2015, 2020-02-24:
> It seems correct except for jumping of the trajectory.

I have tried several times on the KITTI00 sequence. When I ran with the monocular mode, the generated trajectory was plotted as:
x-special/nautilus-clipboard
copy
file:///home/ubuntu/Downloads/plot/KITTI%20results/KITTI00/mono.png

Link: https://pan.baidu.com/s/1r2EWqebeHLMyLeyqCz20jg 
Request code: b2b4
When I ran with the stereo mode, the generated trajectory was plotted as:
x-special/nautilus-clipboard
copy
file:///home/ubuntu/Downloads/plot/KITTI%20results/KITTI00/stereo.png

Link: https://pan.baidu.com/s/1OnE3YuqZpQgmuvXaPJsP-w 
Request code: sg9e

They all seems incorrect with the ground truth reference. OpenvSLAM should estimate more accurate trajectory when comparing with ORB-SLAM, right? Since it is the upgraded version. Did I use the wrong yaml file? Or is there anything I have done incorrectly?

squaresoft2015, 2020-03-05:
> @squaresoft2015
> Trajectories plotted by `keyframe_trajectory.txt` are the same with the ones by `frame_trajectory.txt`?

The frame trajectory and keyframe trajectory are almost the same:
mono link: https://pan.baidu.com/s/1Dw7m2mj4_HrQDBRJg-Pg9Q 
Request code: 5gp0
stereo link: https://pan.baidu.com/s/1bciCzUn_fQ0nyp-1CJkerg 
Request code: suz5

TaylorGy, 2020-04-11:
> > @IaroslavS frame_trajectory.txt contains timestamp, 3dof pose, orientation as Quaternion. you can compute SE3 matrix from the 3dof pose (x,y,z) and orientation-->SO3.
> 
> Thanks.
> There's a mistake in repository in file run_kitti_slam.cc - there should be "KITTI" instead of "TUM"

@IaroslavS Thank you that you've solved my questions. I fixed the run_kitti_slam.cc file by replacing "TUM" with "KITTI", then rebuilt the whole project. Now the program works fine and generates trajectories in KITTI format.

squaresoft2015, 2020-04-12:
@IaroslavS Do you mean the command should be kitti rather than tum? I remember all the generated trajectories are in TUM format.

IaroslavS, 2020-04-21:
> @IaroslavS Do you mean the command should be kitti rather than tum? I remember all the generated trajectories are in TUM format.

There are no special restricions for output format, cause you can always change format for your favor by yourself. But if you're going to upload results to KITTI Benchmark site then output format should be KITTI. 
Script "run_kitti_slam.cc" says that it's for kitti and I intuitively think that it should give out in kitti format.

closed 2020-04-21 by IaroslavS

aiyanxiao, 2020-06-01:
> @squaresoft2015 Trajectory files created by OpenVSLAM are in the TUM-style.
> So first you should convert `frame_trajectory.txt` to the KITTI-style, ant then compare it with the official GT using [...]

Can you help me what to do to convert `frame_trajectory.txt` to the KITTI-style?

aiyanxiao, 2020-06-01:
Can you help me with that what I should do to convert frame_trajectory.txt to the KITTI-style?

IaroslavS, 2020-06-01:
> Can you help me with that what I should do to convert frame_trajectory.txt to the KITTI-style?

cd "your working folder, where you want to have a new repo EVO"
`git clone https://github.com/MichaelGrupp/evo.git
`

`cd evo
`

Make a new python script by creating a new empty file, for example "convert_tum_to_kitti.py" and copy text below into the new made file

```
from evo.tools import file_interface

if __name__ == "__main__":
    import argparse
    parser = argparse.ArgumentParser(description=DESC)
    parser.add_argument("tum_poses", help="input pose path file in TUM format")
    parser.add_argument("kitti_poses", help="output file path for trajectory in KITTI format")
    args = parser.parse_args()
    pose_path = file_interface.read_tum_trajectory_file(args.tum_poses)
    file_interface.write_kitti_poses_file(args.kitti_poses, pose_path)
```

Try to launch this script and don't forget to specify arguments "tum_poses", "kitti_poses" when launching.
This easy script I've written just now in 15 minutes and didn't test it. So if you'll have problems - write here.