xdspacelab/openvslam #245: Sample not working (windows, docker)

valeriob, 2020-02-10:
Hi,
i followed the guide to run the first sample on docker (i'm on windows). 
https://openvslam.readthedocs.io/en/master/simple_tutorial.html#chapter-simple-tutorial

when i run _run_video_slam_ it looks like hanging because nothing shows up in the web interface
[image]

If i press "terminate" in the web interface, a small file map.msg gets created (just 251 bytes)
When i run _run_video_localization_ the video appears in the web interface but the localization is stuck at the starting poing.

[image]

Thanks

minhpvo, 2020-05-15:
I have a similar problem where I built on WSL and used socket to visualize the example in the tutorial.  Nothing shows up in the view. Have folks figured out what happened here?

Here is a screenshot: 
[image]

jchrisli, 2020-07-04:
@valeriob I don't think your slam program was able to connect to the web server, which host the user interfaces. This seems to be due to limitations of Docker through WSL. I was able to work around it by binding the server to '0.0.0.0' rather than localhost in viewer/app.js.
I am using WSL2, not sure if it works for WSL1.