xdspacelab/openvslam #246: Segmentation fault (core dumped)

squaresoft2015, 2020-02-11:
The compilation is completed with Pangolin, but the program stucks at DESTRUCT: initialize::bearing_vector during running. No matter with simple tutorial on MP4 video file or public dataset. --debug output is:

[2020-02-11 10:05:51.676] [I] loading ORB vocabulary: orb_vocab/orb_vocab.dbow2
[2020-02-11 10:05:52.093] [D] CONSTRUCT: data::camera_database
[2020-02-11 10:05:52.093] [D] CONSTRUCT: data::map_database
[2020-02-11 10:05:52.093] [D] CONSTRUCT: data::bow_database
[2020-02-11 10:05:52.093] [D] CONSTRUCT: publish::frame_publisher
[2020-02-11 10:05:52.093] [D] CONSTRUCT: publish::map_publisher
[2020-02-11 10:05:52.093] [D] CONSTRUCT: module::initializer
[2020-02-11 10:05:52.093] [D] CONSTRUCT: module::relocalizer
[2020-02-11 10:05:52.093] [D] CONSTRUCT: tracking_module
[2020-02-11 10:05:52.093] [D] CONSTRUCT: mapping_module
[2020-02-11 10:05:52.093] [D] CONSTRUCT: global_optimization_module
[2020-02-11 10:05:52.093] [I] startup SLAM system
[2020-02-11 10:05:52.093] [I] start mapping module
[2020-02-11 10:05:52.094] [I] start global optimization module
[2020-02-11 10:05:52.388] [D] CONSTRUCT: initialize::bearing_vector
[2020-02-11 10:05:52.637] [I] initialization succeeded with E
[2020-02-11 10:05:52.637] [D] DESTRUCT: initialize::bearing_vector
*** Aborted at 1581386752 (unix time) try "date -d @1581386752" if you are using GNU date ***
PC: @     0x7f7c5395e8a0 cfree
*** SIGSEGV (@0x6d) received by PID 19401 (TID 0x7f7c24728700) from PID 109; stack trace: ***
    @     0x7f7c5495b641 (unknown)
    @     0x7f7c546b2540 (unknown)
    @     0x7f7c5395e8a0 cfree
    @     0x7f7c52d853a5 g2o::JacobianWorkspace::~JacobianWorkspace()
    @     0x7f7c54826a0c g2o::BlockSolver<>::buildSystem()
    @     0x7f7c52d80219 g2o::OptimizationAlgorithmLevenberg::solve()
    @     0x7f7c52d7890c g2o::SparseOptimizer::optimize()
    @     0x7f7c548642f1 openvslam::optimize::global_bundle_adjuster::optimize()
    @     0x7f7c54814257 openvslam::module::initializer::create_map_for_monocular()
    @     0x7f7c548147bb openvslam::module::initializer::initialize()
    @     0x7f7c54784bb0 openvslam::tracking_module::initialize()
    @     0x7f7c54787725 openvslam::tracking_module::track()
    @     0x7f7c54787f71 openvslam::tracking_module::track_monocular_image()
    @     0x7f7c54778821 openvslam::system::feed_monocular_frame()
    @     0x555fa6e6aa31 _ZNSt6thread11_State_implINS_8_InvokerISt5tupleIJZ13mono_trackingRKSt10shared_ptrIN9openvslam6configEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESG_SG_jbbbSG_EUlvE_EEEEE6_M_runEv
    @     0x7f7c53ba4f74 (unknown)
    @     0x7f7c546a6669 start_thread
    @     0x7f7c539e3323 clone
Segmentation fault (core dumped)

Any help? It seems like a error caused by empty pointer. I have configured and rebuilded g2o and OpenVSLAM with -DBUILD_WITH_MARCH_NATIVE=OFF option for cmake. But the error remains.

closed 2020-02-21 by squaresoft2015

RahulSurana123, 2020-03-16:
How did you solve this issue ??

dannyway03, 2020-03-18:
Hi, I'm facing the exact same issue. Can you provide guidance on how to fix it?

squaresoft2015, 2020-03-18:
Actually I just changed to a totally new Ubuntu system and reinstalled everything. The error does not exist and I don't know why.

RahulSurana123, 2020-03-18:
use just need to delete your g2o folder and build g2o from scratch to make it work

dannyway03, 2020-03-19:
thanks a lot! that solved the problem