xdspacelab/openvslam #247: Openvslam and ROS2

mirellameelo, 2020-02-14:
Hi, nice job! 

Do you have any version working with ros2?  Does anyone has done it? Tks!

klintan, 2020-02-19:
Hey, 
Checkout this! https://github.com/xdspacelab/openvslam/pull/135 It should be finished, however not merged yet. If you encounter any issues, feel free to report them so that we can fix them :)

mirellameelo, 2020-02-19:
Hi, @klintan! Tks. :)

I was trying it, but i`m little confuse in how to run it. Let me tell you my step-by-step:

1) I installed all [dependencies](https://openvslam.readthedocs.io/en/master/installation.html) of openvSLAM;
2) I clone the [ros2 branch](https://github.com/klintan/openvslam/tree/ros2) and followed the build instructions from the official documentation;
3) To build the packages, i runnes in the the terminal:
```sh
source /opt/ros/dashing/setup.bash
cd /path/to/opencvslam 
colcon build
```
I had an issue: CXSparse was NOT found, so I installed CeresSolver:

```sh
git clone https://ceres-solver.googlesource.com/ceres-solver
```

Then,
```sh
colcon build 
. install/setup.bash
```
The package apparently, was built. But I can`t find it... Did I make any mistakes in my steps? For now, I`m looking for a tutorial about how to [creat packages in ros2](https://index.ros.org/doc/ros2/Tutorials/Writing-A-Simple-Cpp-Publisher-And-Subscriber/) to find answers. 

I appreciate any help. Tks. :)

klintan, 2020-02-22:
Hm that is stranged, I tried it now from scratch, and it worked for me (assuming you have all openvslam dependencies installed):

1. `git clone https://github.com/klintan/openvslam.git` 
2. `git checkout ros2`
3. Activate/source your ros2 environment
4. `mkdir build && cd build`
5. ```
cmake     -DBUILD_WITH_MARCH_NATIVE=ON     -DUSE_PANGOLIN_VIEWER=ON     -DUSE_SOCKET_PUBLISHER=OFF     -DUSE_STACK_TRACE_LOGGER=ON     -DBOW_FRAMEWORK=DBoW2     -DBUILD_TESTS=ON     ..
```
6. `make -j4`
7. `make install`
8. `cd .. & cd ros2`

Before step 9, it assumes you have `image_common` and `vision_opencv` libraries sourced in your ros environment, if not you'll have to source them before running step 9.

9. ```
colcon build --cmake-args -DBUILD_WITH_MARCH_NATIVE=ON     -DUSE_PANGOLIN_VIEWER=ON     -DUSE_SOCKET_PUBLISHER=OFF     -DUSE_STACK_TRACE_LOGGER=ON     -DBOW_FRAMEWORK=DBoW2
```
10. Activate/source openvslam/ros2 environment (`source ./install/setup.bash`) 

Doing all this I see 
```
openvslam run_localization
openvslam run_slam
```

when running `ros2 pkg executables`.

Let me know if it works for you.

mirellameelo, 2020-02-26:
@klintan It works! Tks so much!

I just changed your step 9 for:

```bh
colcon build --symplink-install
```

So, I did a README with the entire step-by-step. Would be nice if you could test it and add it to your repository. Tks again!!

# README file:

## Dependencies

### Image common packages

1) Source the ros2 setup files

```bh
    source /opt/ros/<DISTRO>/setup.bash
```
2) Build the image_common packages

```bh
    mkdir $HOME/ws && cd $HOME/ws
    git clone -b <DISTRO> --single-branch https://github.com/ros-perception/image_common.git
    cd image_common 
    colcon build --symlink-install
```

It should build 3 packs:
- camera_calibration_parsers
- image_transport
- camera_info_manager

### Vision openCV packages

3) Build the vision_opencv packages 

```bh
    cd $HOME/ws
    git clone -b ros2 --single-branch https://github.com/ros-perception/vision_opencv.git
    cd vision_opencv 
    colcon build --symlink-install
```

It should build 4 packs:
- cv_bridge
- image_geometry
- opencv_tests
- vision_opencv

### Openvslam packages

4. Source image_common and vision_opencv setup files

```bh
    source $HOME/ws/vision_opencv/install/setup.bash
    source $HOME/ws/image_common/install/setup.bash
```

5. Build openvslam packages

```
    cd $HOME
    git clone -b ros2 --single-branch https://github.com/klintan/openvslam.git
    cd openvslam
    mkdir build && cd build
    cmake \
        -DBUILD_WITH_MARCH_NATIVE=ON \
        -DUSE_PANGOLIN_VIEWER=ON \
        - DUSE_SOCKET_PUBLISHER=OFF \     
        -DUSE_STACK_TRACE_LOGGER=ON \    
        -DBOW_FRAMEWORK=DBoW2 \
        -DBUILD_TESTS=ON \    
        ..
    make -j4
    sudo make install
    cd ../ros2
    colcon build --symlink-install
```

It should build 2 packs:
- openvslam
- publisher

# Runnning with a dataset 

1. Download:
    a. video.mp4 file
    b. config.yaml file
    c. vocabulary.dbow2 file

```bh
    cd $HOME/openvslam/build
    
    # video and config files
    FILE_ID="1d8kADKWBptEqTF7jEVhKatBEdN7g0ikY"
    curl -sc /tmp/cookie "https://drive.google.com/uc?export=download&id=${FILE_ID}" > /dev/null
    CODE="$(awk '/_warning_/ {print $NF}' /tmp/cookie)"
    curl -sLb /tmp/cookie "https://drive.google.com/uc?export=download&confirm=${CODE}&id=${FILE_ID}" -o aist_living_lab_1.zip
    unzip aist_living_lab_1.zip
    
    # vocabulary
    FILE_ID="1wUPb328th8bUqhOk-i8xllt5mgRW4n84"
    curl -sc /tmp/cookie "https://drive.google.com/uc?export=download&id=${FILE_ID}" > /dev/null
    CODE="$(awk '/_warning_/ {print $NF}' /tmp/cookie)"
    curl -sLb /tmp/cookie "https://drive.google.com/uc?export=download&confirm=${CODE}&id=${FILE_ID}" -o orb_vocab.zip
    unzip orb_vocab.zip
```

2. Open 3 terminals and source vision_opencv, image_common and openvslam setup files in each terminal:

```bh
    source $HOME/ws/vision_opencv/install/setup.bash
    source $HOME/ws/image_common/install/setup.bash
    source $HOME/openvslam/ros2/install/setup.bash
```

Terminal 1: publish the video
```bh
    ros2 run publisher video -m $HOME/openvslam/build/aist_living_lab_1/video.mp4
```

Terminal 2: republish the video
```bh
    ros2 run image_transport republish raw in:=/video/image_raw raw out:=/camera/image_raw
```

Terminal 3: run_slam mode
```bh
    ros2 run openvslam run_slam -v $HOME/openvslam/build/orb_vocab/orb_vocab.dbow2 -c             $HOME/openvslam/build/aist_living_lab_1/config.yaml 
```

closed 2020-05-17 by mirellameelo