xdspacelab/openvslam #248: problems in CMake lists

Mikhail2017, 2020-02-16:
Hi
I'm trying to build on Ubuntu 18.04.
Have created build directory in cloned openvslam folder and succesfully configured a make project. Unfortunately I've got a lot of include errors like this one:

system.cc:1:10: fatal error: openvslam/system.h: No such file or directory
 #include "openvslam/system.h"

Looks like a problem in build scripts

shinsumicco, 2020-02-17:
@Mikhail2017 
Could you tell us the more details about the commands you used? (from `git clone` to `cmake`)

Mikhail2017, 2020-02-17:
```
git clone https://github.com/xdspacelab/openvslam
cd openvslam
mkdir build
cd build
cmake .. -DCMAKE_TOOLCHAIN_FILE=~/Coding/vcpkg/scripts/buildsystems/vcpkg.cmake -DUSE_PANGOLIN_VIEWER=OFF -DBUILD_WITH_MARCH_NATIVE=ON -DUSE_SOCKET_PUBLISHER=ON -DUSE_STACK_TRACE_LOGGER=ON -DBOW_FRAMEWORK=DBoW2 -DBUILD_TESTS=ON
make -j4
```

All deps already installed using vcpkg

lessthantrue, 2020-09-17:
I'm having this same problem. Has anyone found a solution? I'm not experienced enough with Cmake to figure it out myself.