xdspacelab/openvslam #249: No mapping result when running the run_video_slam example

D7823, 2020-02-18:
Hi,
 I calibrated a pi camera and recorded a short video(around 1min) to have a quick test with the run_video_slam application under the example folder. I fed the video, camera config file, and vocabulary file to start the SLAM process. But the map is unable to create (no point cloud and the position of the camera is not moving)  and I notice there is not much yellow point and lines shown in the video. Can I directly use that code for testing or I need to change something? Or Is that too little extracted orb features? If so, how should I adjust the feature parameters to get a better result?

Thanks in advance

reunanen, 2020-02-18:
I had a similar issue. Adding something like this in the config yaml file helped in my case:

```
Initializer.num_min_triangulated_pts: 5
Initializer.parallax_deg_threshold: 0.8
```

D7823, 2020-02-19:
Thank you. Those initializer parameters didn't help much in my case. But I recorded another video and it worked okay so it did because not much keypoints got extracted from the frames.

closed 2020-02-19 by D7823