xdspacelab/openvslam #252: Build doesnt work for CMakeList.txt

pramo23, 2020-02-20:
basically i followed the installation steps given on Installation step on my new virtual machine in Ubuntu 16.04 

I got following error
[ 66%] Completed 'googletest-download'
[100%] Built target googletest-download
CMake Error at build/googletest-src/CMakeLists.txt:13 (if):
  if given arguments:

    "CMAKE_VERSION" "VERSION_GREATER_EQUAL" "3.0.2"

  Unknown arguments specified

Similar issue is created on https://github.com/google/googletest/issues/2711
but it wasnt helpful in my case 
Anyone else facing the same problem ?

Thank you.

shinsumicco, 2020-02-21:
@pramo23 
Thank you for the report.
The CMake error you reported is caused because CMake < v3.8 does not have the control statement `VERSION_GREATER_EQUAL`.
It seems that `CMakeLists.txt` of google-test was updated recently. As a result of it, the error message has become to appeared.

I fixed the version of google-test, whose `CMakeLists.txt` does not have `VERSION_GREATER_EQUAL` statement.

closed 2020-02-21 by shinsumicco

shinsumicco, 2020-02-22:
@pramo23 Please try the latest `master` branch.

pramo23, 2020-02-24:
@shinsumicco thank you very much. It working now :)

PR #254 (2020-02-21): Fix the version of google-test
fixed #252