xdspacelab/openvslam #253: Error building https://github.com/shinsumicco/DBoW2.git

dagon0, 2020-02-21:
Hi,
when i try to build DBoW2 i get a linking problem (or at least i think it's about linking)

OS: Ubuntu 18.04

First i do:
```
cd /opt
git clone https://github.com/shinsumicco/DBoW2.git
cd DBoW2
mkdir build && cd build
cmake \
    -DCMAKE_BUILD_TYPE=Release \
    -DCMAKE_INSTALL_PREFIX=/usr/local \
    ..
```

And this is the output:
```

-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Use OpenCV 4.1.0
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/DBoW2/build
```

Then i do:

`make -j4`

And i get these errors:

```
Scanning dependencies of target DBoW2
[ 22%] Building CXX object CMakeFiles/DBoW2.dir/src/BowVector.cpp.o
[ 22%] Building CXX object CMakeFiles/DBoW2.dir/src/FeatureVector.cpp.o
[ 33%] Building CXX object CMakeFiles/DBoW2.dir/src/QueryResults.cpp.o
[ 44%] Building CXX object CMakeFiles/DBoW2.dir/src/ScoringObject.cpp.o
[ 55%] Building CXX object CMakeFiles/DBoW2.dir/src/FORB.cpp.o
[ 66%] Building CXX object CMakeFiles/DBoW2.dir/src/FBrief.cpp.o
[ 77%] Linking CXX shared library libDBoW2.so
[ 77%] Built target DBoW2
Scanning dependencies of target convert_orb_vocab
[ 88%] Building CXX object CMakeFiles/convert_orb_vocab.dir/util/convert_orb_vocab.cpp.o
[100%] Linking CXX executable convert_orb_vocab
CMakeFiles/convert_orb_vocab.dir/util/convert_orb_vocab.cpp.o: In function `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > cv::CommandLineParser::get<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) const':
convert_orb_vocab.cpp:(.text._ZNK2cv17CommandLineParser3getINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEET_RKS7_b[_ZNK2cv17CommandLineParser3getINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEET_RKS7_b]+0x2c): undefined reference to `cv::CommandLineParser::getByName(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, cv::Param, void*) const'
CMakeFiles/convert_orb_vocab.dir/util/convert_orb_vocab.cpp.o: In function `DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>::load(cv::FileStorage const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
convert_orb_vocab.cpp:(.text._ZN5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4loadERKNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4loadERKNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x148): undefined reference to `cv::FileStorage::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
convert_orb_vocab.cpp:(.text._ZN5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4loadERKNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4loadERKNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x3f5): undefined reference to `cv::FileNode::string[abi:cxx11]() const'
CMakeFiles/convert_orb_vocab.dir/util/convert_orb_vocab.cpp.o: In function `main':
convert_orb_vocab.cpp:(.text.startup+0x51): undefined reference to `cv::CommandLineParser::CommandLineParser(int, char const* const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
convert_orb_vocab.cpp:(.text.startup+0xb9): undefined reference to `cv::CommandLineParser::has(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
convert_orb_vocab.cpp:(.text.startup+0x128): undefined reference to `cv::CommandLineParser::has(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
convert_orb_vocab.cpp:(.text.startup+0x17f): undefined reference to `cv::CommandLineParser::has(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
convert_orb_vocab.cpp:(.text.startup+0x50a): undefined reference to `cv::CommandLineParser::has(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
[... 26 lines cut ...]
convert_orb_vocab.cpp:(.text._ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x77e): undefined reference to `cv::operator<<(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
CMakeFiles/convert_orb_vocab.dir/util/convert_orb_vocab.cpp.o:convert_orb_vocab.cpp:(.text._ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x7c9): more undefined references to `cv::operator<<(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)' follow
CMakeFiles/convert_orb_vocab.dir/util/convert_orb_vocab.cpp.o: In function `DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>::save(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const':
convert_orb_vocab.cpp:(.text._ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x8d5): undefined reference to `cv::write(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
convert_orb_vocab.cpp:(.text._ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x91a): undefined reference to `cv::operator<<(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
convert_orb_vocab.cpp:(.text._ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x95d): undefined reference to `cv::write(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
convert_orb_vocab.cpp:(.text._ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x998): undefined reference to `cv::operator<<(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
convert_orb_vocab.cpp:(.text._ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x9ec): undefined reference to `cv::operator<<(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
convert_orb_vocab.cpp:(.text._ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0xa32): undefined reference to ` [... cut ...] >, std::allocator<char> > const&, char const*, char const*, int)' follow
collect2: error: ld returned 1 exit status
CMakeFiles/convert_orb_vocab.dir/build.make:110: recipe for target 'convert_orb_vocab' failed
make[2]: *** [convert_orb_vocab] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/convert_orb_vocab.dir/all' failed
make[1]: *** [CMakeFiles/convert_orb_vocab.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

Thank you a lot in advance

shinsumicco, 2020-02-21:
Hmm... I could not reproduce it.
There seems some reasons for the link error:

- the old DBoW2 library remains anywhere in your environment
- Some OpenCV modules are broken, or not placed at proper directory.

If you want to use only the DBoW2 lib (`libDBoW2.so`), you can do it with `cmake .. -DBUILD_UTIL=OFF` configuration.

dagon0, 2020-02-21:
Thanks a lot for answering so fast.
Actually at the moment i would like to install this custom version of DBoW2 to subsequently build and install openvslam.
When i try to build and install openvslam using these:

```
cd /opt/openvslam
mkdir build && cd build
cmake \
    -DBUILD_WITH_MARCH_NATIVE=ON \
    -DUSE_PANGOLIN_VIEWER=OFF \
    -DUSE_SOCKET_PUBLISHER=ON \
    -DUSE_STACK_TRACE_LOGGER=ON \
    -DBOW_FRAMEWORK=DBoW2 \
    -DBUILD_TESTS=ON \
    ..
```

Cmake doesn't finish and gives this error:

```
-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type: Release
-- ccache: Disabled
-- Address/Memory sanitizer: DISABLED
-- Architecture-aware optimization (-march=native): ENABLED
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Eigen3: /usr/local/include/eigen3 (Required is at least version "3.3") 
-- Found OpenCV: /usr/local (found suitable version "4.1.0", minimum required is "4.0") found components:  core imgcodecs videoio features2d calib3d highgui 
-- Use OpenCV 4.1.0
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Failed to find CXSparse - Could not find CXSparse include directory, set CXSPARSE_INCLUDE_DIR to directory containing cs.h
-- A library with BLAS API found.
-- Found AMD library: /usr/lib/x86_64-linux-gnu/libamd.so
-- Found AMD header in: /usr/include/suitesparse
-- Found CAMD library: /usr/lib/x86_64-linux-gnu/libcamd.so
-- Found CAMD header in: /usr/include/suitesparse
-- Found COLAMD library: /usr/lib/x86_64-linux-gnu/libcolamd.so
-- Found COLAMD header in: /usr/include/suitesparse
-- Found CCOLAMD library: /usr/lib/x86_64-linux-gnu/libccolamd.so
-- Found CCOLAMD header in: /usr/include/suitesparse
-- Found SuiteSparse_config library: /usr/lib/x86_64-linux-gnu/libsuitesparseconfig.so
-- Found SuiteSparse_config header in: /usr/include/suitesparse
-- Adding librt: /usr/lib/x86_64-linux-gnu/librt.so to SuiteSparse_config libraries (required on Linux & Unix [not OSX] if SuiteSparse is compiled with timing).
-- Did not find METIS library (optional SuiteSparse dependency)
-- Found SuiteSparse: TRUE (found version "5.1.2") 
-- OpenMP: DISABLED
-- SSE3 for ORB extraction (-msse3): DISABLED
-- SSE for floating-point operation (-mfpmath=sse): DISABLED
CMake Error at src/openvslam/CMakeLists.txt:101 (find_package):
  By not providing "FindDBoW2.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "DBoW2", but
  CMake did not find one.

  Could not find a package configuration file provided by "DBoW2" with any of
  the following names:

    DBoW2Config.cmake
    dbow2-config.cmake

  Add the installation prefix of "DBoW2" to CMAKE_PREFIX_PATH or set
  "DBoW2_DIR" to a directory containing one of the above files.  If "DBoW2"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/opt/openvslam/build/CMakeFiles/CMakeOutput.log".
See also "/opt/openvslam/build/CMakeFiles/CMakeError.log".

```

So i tried to fix the DBoW2 installation but it gives me that error that i pasted in the last post.

You are right about the old DBoW2 library that remains the environment, i installed in the past the old one to install Opencv

What do you suggest me to do in order to build Openvslam? Should i clean all the libraries installations (Opencv, Dbow2, G2o...)? If so, how could i do that in the right way?

Thanks again for your time, i really appreciate it

shinsumicco, 2020-02-22:
Please try to specify the env `DBoW2_DIR` so that `cmake` can find CMake configurations of DBoW2. 
For example, if you configure `cmake` without any `-DCMAKE_INSTALL_PREFIX=/path/to/install` option when building and installing DBoW2 , please execute
```sh
# check if the CMake configurations are placed at the proper path
$ ls /usr/local/lib/cmake/DBoW2
DBoW2Config.cmake
# specify the env
$ export DBoW2_DIR=/usr/local/lib/cmake/DBoW2
```
before running `cmake` of OpenVSLAM.

dagon0, 2020-02-22:
The folder /usr/local/lib/cmake/DBoW2 didin't exist, so i created it and i copied the contents of /opt/DBoW2/build/ to /usr/local/lib/cmake/DBoW2:

```
ls /usr/local/lib/cmake/DBoW2

CMakeCache.txt
CMakeFiles
cmake_install.cmake
DBoW2Config.cmake
libDBoW2.so
Makefile

```
and i set the env variable:

`export DBoW2_DIR=/usr/local/lib/cmake/DBoW2`

Then i tried to build OpenVSLAM again:

```
cd /opt/openvslam
mkdir build && cd build
cmake \
    -DBUILD_WITH_MARCH_NATIVE=ON \
    -DUSE_PANGOLIN_VIEWER=OFF \
    -DUSE_SOCKET_PUBLISHER=ON \
    -DUSE_STACK_TRACE_LOGGER=ON \
    -DBOW_FRAMEWORK=DBoW2 \
    -DBUILD_TESTS=ON \
    ..
```
And this time it generates the build files! :)
(Also if there is a recurrent error) 

```
-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type: Release
-- ccache: Disabled
-- Address/Memory sanitizer: DISABLED
-- Architecture-aware optimization (-march=native): ENABLED
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Eigen3: /usr/local/include/eigen3 (Required is at least version "3.3") 
-- Found OpenCV: /usr/local (found suitable version "4.1.0", minimum required is "4.0") found components:  core imgcodecs videoio features2d calib3d highgui 
-- Use OpenCV 4.1.0
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Failed to find CXSparse - Could not find CXSparse include directory, set CXSPARSE_INCLUDE_DIR to directory containing cs.h
-- A library with BLAS API found.
-- Found AMD library: /usr/lib/x86_64-linux-gnu/libamd.so
-- Found AMD header in: /usr/include/suitesparse
[... 315 lines cut ...]
Scanning dependencies of target test_helper
Scanning dependencies of target gtest
Scanning dependencies of target openvslam
[  1%] Building CXX object googletest-build/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[  2%] Building CXX object test/helper/CMakeFiles/test_helper.dir/keypoint.cc.o
[  3%] Building CXX object test/helper/CMakeFiles/test_helper.dir/bearing_vector.cc.o
[  4%] Building CXX object src/openvslam/CMakeFiles/openvslam.dir/config.cc.o
/opt/openvslam/src/openvslam/config.cc:1:10: fatal error: openvslam/config.h: No such file or directory
 #include "openvslam/config.h"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
src/openvslam/CMakeFiles/openvslam.dir/build.make:62: recipe for target 'src/openvslam/CMakeFiles/openvslam.dir/config.cc.o' failed
make[2]: *** [src/openvslam/CMakeFiles/openvslam.dir/config.cc.o] Error 1
CMakeFiles/Makefile2:138: recipe for target 'src/openvslam/CMakeFiles/openvslam.dir/all' failed
make[1]: *** [src/openvslam/CMakeFiles/openvslam.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  4%] Building CXX object test/helper/CMakeFiles/test_helper.dir/landmark.cc.o
[  5%] Linking CXX shared library libtest_helper.so
[  5%] Built target test_helper
[  6%] Linking CXX static library ../../lib/libgtest.a
[  6%] Built target gtest
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

```

shinsumicco, 2020-02-22:
@dagon0 
Did you execute `make install` after building `DBoW2`?
If installed, a CMake configuration file for DBoW2 must be **automatically** placed under `/usr/local/lib/cmake/DBoW2`.
(Note: If you specified the `CMAKE_INSTALL_PREFIX` option when executing `cmake`, the directory for it is decided according to the option.)

dagon0, 2020-02-24:
I didn't execute `make install` after building DBoW2,
When i try now to execute it i get this error:
```
[ 77%] Built target DBoW2
[ 88%] Linking CXX executable convert_orb_vocab
CMakeFiles/convert_orb_vocab.dir/util/convert_orb_vocab.cpp.o: In function `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > cv::CommandLineParser::get<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) const':
convert_orb_vocab.cpp:(.text._ZNK2cv17CommandLineParser3getINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEET_RKS7_b[_ZNK2cv17CommandLineParser3getINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEET_RKS7_b]+0x2c): undefined reference to `cv::CommandLineParser::getByName(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, cv::Param, void*) const'
CMakeFiles/convert_orb_vocab.dir/util/convert_orb_vocab.cpp.o: In function `DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>::load(cv::FileStorage const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
convert_orb_vocab.cpp:(.text._ZN5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4loadERKNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4loadERKNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x148): undefined reference to `cv::FileStorage::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
convert_orb_vocab.cpp:(.text._ZN5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4loadERKNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4loadERKNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x3f5): undefined reference to `cv::FileNode::string[abi:cxx11]() const'
CMakeFiles/convert_orb_vocab.dir/util/convert_orb_vocab.cpp.o: In function `main':
convert_orb_vocab.cpp:(.text.startup+0x51): undefined reference to `cv::CommandLineParser::CommandLineParser(int, char const* const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
convert_orb_vocab.cpp:(.text.startup+0xb9): undefined reference to `cv::CommandLineParser::has(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
convert_orb_vocab.cpp:(.text.startup+0x128): undefined reference to `cv::CommandLineParser::has(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
convert_orb_vocab.cpp:(.text.startup+0x17f): undefined reference to `cv::CommandLineParser::has(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
convert_orb_vocab.cpp:(.text.startup+0x50a): undefined reference to `cv::CommandLineParser::has(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
convert_orb_vocab.cpp:(.text.startup+0x5b0): undefined reference to `cv::CommandLineParser::has(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
CMakeFiles/convert_orb_vocab.dir/util/convert_orb_vocab.cpp.o:convert_orb_vocab.cpp:(.text.startup+0x5f1): more undefined references to `cv::CommandLineParser::has(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const' follow
CMakeFiles/convert_orb_vocab.dir/util/convert_orb_vocab.cpp.o: In function `DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>::save(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const':
convert_orb_vocab.cpp:(.text._ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x3b): undefined reference to `cv::operator<<(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
convert_orb_vocab.cpp:(.text._ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x6b): undefined reference to `cv::operator<<(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
convert_orb_vocab.cpp:(.text._ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0xb1): undefined reference to `cv::operator<<(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
convert_orb_vocab.cpp:(.text._ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZNK5DBoW219TemplatedVocabularyIN2cv3MatENS_4FORBEE4saveERNS1_11FileStorageERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0xf5): undefined reference to `cv::write(cv::FileStorage&, std::__cxx11::basic_string<char, std::c [... cut ...]
make[1]: *** [CMakeFiles/convert_orb_vocab.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

I also noticed that there's something wrong with my OpenCV installation:
```
python
Python 3.6.9 (default, Nov  7 2019, 10:44:02) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: libz-a147dcb0.so.1.2.3: cannot open shared object file: No such file or directory
>>> 
```

shinsumicco, 2020-02-24:
@dagon0 
Please try to execute the following commands one by one:
```
cmake .. -DBUILD_UTIL=OFF
make -j4
sudo make install
```

dagon0, 2020-02-24:
Thanks! :) it works now:

cmake .. -DBUILD_UTIL=OFF

-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Use OpenCV 4.1.0
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/DBoW2/build

make -j4

[ 28%] Building CXX object CMakeFiles/DBoW2.dir/src/FeatureVector.cpp.o
[ 28%] Building CXX object CMakeFiles/DBoW2.dir/src/QueryResults.cpp.o
[ 42%] Building CXX object CMakeFiles/DBoW2.dir/src/BowVector.cpp.o
[ 57%] Building CXX object CMakeFiles/DBoW2.dir/src/ScoringObject.cpp.o
[ 71%] Building CXX object CMakeFiles/DBoW2.dir/src/FORB.cpp.o
[ 85%] Building CXX object CMakeFiles/DBoW2.dir/src/FBrief.cpp.o
[100%] Linking CXX shared library libDBoW2.so
[100%] Built target DBoW2

make install

[100%] Built target DBoW2
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/include/DBoW2
-- Installing: /usr/local/include/DBoW2/FBrief.h
-- Installing: /usr/local/include/DBoW2/TemplatedVocabulary.h
-- Installing: /usr/local/include/DBoW2/DBoW2.h
-- Installing: /usr/local/include/DBoW2/BowVector.h
-- Installing: /usr/local/include/DBoW2/TemplatedDatabase.h
-- Installing: /usr/local/include/DBoW2/ScoringObject.h
-- Installing: /usr/local/include/DBoW2/FeatureVector.h
-- Installing: /usr/local/include/DBoW2/QueryResults.h
-- Installing: /usr/local/include/DBoW2/FClass.h
-- Installing: /usr/local/include/DBoW2/FSurf64.h
-- Installing: /usr/local/include/DBoW2/FORB.h
-- Installing: /usr/local/lib/libDBoW2.so
-- Set runtime path of "/usr/local/lib/libDBoW2.so" to ""
-- Installing: /usr/local/lib/cmake/DBoW2/DBoW2Config.cmake

dagon0, 2020-02-24:
Thank you so much for the help, i'm so happy that it worked.

After that i tried to build and install OpenVSLAM, so i first executed the cmake without any problems (at least it seems so to me):

```
cmake \
    -DBUILD_WITH_MARCH_NATIVE=ON \
    -DUSE_PANGOLIN_VIEWER=OFF \
    -DUSE_SOCKET_PUBLISHER=ON \
    -DUSE_STACK_TRACE_LOGGER=ON \
    -DBOW_FRAMEWORK=DBoW2 \
    -DBUILD_TESTS=ON \
    ..

-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type: Release
-- ccache: Disabled
-- Address/Memory sanitizer: DISABLED
-- Architecture-aware optimization (-march=native): ENABLED
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Eigen3: /usr/local/include/eigen3 (Required is at least version "3.3") 
-- Found OpenCV: /usr/local (found suitable version "4.1.0", minimum required is "4.0") found components:  core imgcodecs videoio features2d calib3d highgui 
-- Use OpenCV 4.1.0
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Failed to find CXSparse - Could not find CXSparse include directory, set CXSPARSE_INCLUDE_DIR to directory containing cs.h
-- A library with BLAS API found.
-- Found AMD library: /usr/lib/x86_64-linux-gnu/libamd.so
-- Found AMD header in: /usr/include/suitesparse
-- Found CAMD library: /usr/lib/x86_64-linux-gnu/libcamd.so
-- Found CAMD header in: /usr/include/suitesparse
-- Found COLAMD library: /usr/lib/x86_64-linux-gnu/libcolamd.so
-- Found COLAMD header in: /usr/include/suitesparse
-- Found CCOLAMD library: /usr/lib/x86_64-linux-gnu/libccolamd.so
-- Found CCOLAMD header in: /usr/include/suitesparse
-- Found SuiteSparse_config library: /usr/lib/x86_64-linux-gnu/libsuitesparseconfig.so
-- Found SuiteSparse_config header in: /usr/include/suitesparse
-- Adding librt: /usr/lib/x86_64-linux-gnu/librt.so to SuiteSparse_config libraries (required on Linux & Unix [not OSX] if SuiteSparse is compiled with timing).
-- Did not find METIS library (optional SuiteSparse dependency)
-- Found SuiteSparse: TRUE (found version "5.1.2") 
-- OpenMP: DISABLED
-- SSE3 for ORB extraction (-msse3): DISABLED
-- SSE for floating-point operation (-mfpmath=sse): DISABLED
-- BoW framework: DBoW2 (found in /usr/local/include)
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.0.0") 
-- Found protoc executable: /usr/bin/protoc
-- No preference for use of exported glog CMake configuration set, and no hints for include/library directories provided. Defaulting to preferring an installed/exported glog CMake configuration if available.
-- Failed to find installed glog CMake configuration, searching for glog build directories exported with CMake.
-- Failed to find an installed/exported CMake configuration for glog, will perform search for installed glog components.
-- Found Glog: /usr/include  
[... 112 lines cut ...]
[ 52%] Building CXX object src/openvslam/CMakeFiles/openvslam.dir/optimize/g2o/sim3/forward_reproj_edge.cc.o
[ 53%] Building CXX object src/openvslam/CMakeFiles/openvslam.dir/publish/frame_publisher.cc.o
[ 53%] Building CXX object src/openvslam/CMakeFiles/openvslam.dir/publish/map_publisher.cc.o
[ 54%] Building CXX object src/openvslam/CMakeFiles/openvslam.dir/solve/common.cc.o
[ 55%] Building CXX object src/openvslam/CMakeFiles/openvslam.dir/solve/homography_solver.cc.o
[ 55%] Building CXX object src/openvslam/CMakeFiles/openvslam.dir/solve/fundamental_solver.cc.o
[ 56%] Building CXX object src/openvslam/CMakeFiles/openvslam.dir/solve/essential_solver.cc.o
[ 57%] Building CXX object src/openvslam/CMakeFiles/openvslam.dir/solve/pnp_solver.cc.o
[ 58%] Building CXX object src/openvslam/CMakeFiles/openvslam.dir/solve/sim3_solver.cc.o
[ 58%] Building CXX object src/openvslam/CMakeFiles/openvslam.dir/util/converter.cc.o
[ 59%] Building CXX object src/openvslam/CMakeFiles/openvslam.dir/util/image_converter.cc.o
[ 60%] Building CXX object src/openvslam/CMakeFiles/openvslam.dir/util/random_array.cc.o
[ 61%] Building CXX object src/openvslam/CMakeFiles/openvslam.dir/util/stereo_rectifier.cc.o
[ 61%] Linking CXX shared library ../../lib/libopenvslam.so
/usr/bin/ld: /usr/local/lib/libyaml-cpp.a(scanscalar.cpp.o): relocation R_X86_64_PC32 against symbol `_ZNK4YAML5RegEx14MatchUncheckedINS_16StreamCharSourceEEEiRKT_' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
src/openvslam/CMakeFiles/openvslam.dir/build.make:1925: recipe for target 'lib/libopenvslam.so' failed
make[2]: *** [lib/libopenvslam.so] Error 1
CMakeFiles/Makefile2:138: recipe for target 'src/openvslam/CMakeFiles/openvslam.dir/all' failed
make[1]: *** [src/openvslam/CMakeFiles/openvslam.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

```

dagon0, 2020-02-24:
Considering that i can now successfully build and install DBoW2 library, i think i'm gonna close this issue.
I'll do more tests to build OpenVSLAM, now that i can build and install DBoW2.

@shinsumicco Thanks a lot, i really appreciate your help in resolving the problem :)

closed 2020-02-24 by dagon0