xdspacelab/openvslam #261: Is there openvslam cuda version?

JunFang-NWPU, 2020-03-04:
Hi,

The GPU is popular today, e.g., jetson nano, will the openvslam have the cuda version soon, since the  orb-slam2, vins-fusion all have cuda version.

Thanks.

Shashika007, 2020-03-06:
@JunFang-NWPU Still there is no GPU version. But you can follow the works of orb-slam2. I am currently developing opencl support version of openvslam. You can build your opencv with your gpu (cuda/opencl) first. Then, you can give it a try replacing some of the opencv functions that optimized by gpu which are using in openvslam. As an example, resize() or gaussianblur(). Otherwise, you can create a custom kernel for a function which you think that consume performance most.

JunFang-NWPU, 2020-03-07:
OK, thanks.

closed 2020-03-07 by JunFang-NWPU

anakin0620, 2020-06-14:
> Hi,
> 
> The GPU is popular today, e.g., jetson nano, will the openvslam have the cuda version soon, since the orb-slam2, vins-fusion all have cuda version.
> 
> Thanks.

I am also working on this project.

dimaxano, 2020-06-15:
Hi, @anakin0620! Do you need some help with that?

anakin0620, 2020-08-09:
@dimaxano Hi, have you done the work? I remember your ID in ORB_SLAM3 issues ;D

dimaxano, 2020-08-12:
Hi, @anakin0620! No, not yet. Have very low priority on that :(
Also CUDA is not my solid skill, so I can be more helpful as a second pilot to someone who knows this stuff better

surfii3z, 2020-09-15:
> Hi, @anakin0620! No, not yet. Have very low priority on that :(
> Also CUDA is not my solid skill, so I can be more helpful as a second pilot to someone who knows this stuff better

I tried to put together the work of [ORB_SLAM2 CUDA](https://github.com/yunchih/ORB-SLAM2-GPU2016-final) at my [repro.](https://github.com/surfii3z/openvslam/tree/gpu_dev2) It seems like the major change is only in [orb_extractor.cc](https://github.com/surfii3z/openvslam/blob/gpu_dev2/src/openvslam/feature/orb_extractor.cc) file.

However, when I ran the program, I got this error.

```
OpenCV Error: Gpu API call (an illegal memory access was encountered) in download, file /home/jedsadakorn/git/opencv-3.4.0/modules/core/src/cuda/gpu_mat.cu, line 260
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/jedsadakorn/git/opencv-3.4.0/modules/core/src/cuda/gpu_mat.cu:260: error: (-217) an illegal memory access was encountered in function download

*** Aborted at 1600090278 (unix time) try "date -d @1600090278" if you are using GNU date ***
PC: @     0x7f06f039ef47 gsignal
*** SIGABRT (@0x3e8000035ec) received by PID 13804 (TID 0x7f0649fff700) from PID 13804; stack trace: ***
    @     0x7f06f35ea8a0 (unknown)
    @     0x7f06f039ef47 gsignal
    @     0x7f06f03a08b1 abort
    @     0x7f06f09f5957 (unknown)
    @     0x7f06f09fbae6 (unknown)
    @     0x7f06f09fbb21 std::terminate()
    @     0x7f06f09fbd54 __cxa_throw
    @     0x7f06f19c6742 cv::error()
    @     0x7f06f19c689f cv::error()
    @     0x7f06f1a47c99 cv::cuda::GpuMat::download()
    @     0x7f06f3a223aa openvslam::cuda::GpuOrb::join()
    @     0x7f06f391293c openvslam::feature::orb_extractor::extract()
    @     0x7f06f38ddb0a openvslam::data::frame::extract_orb()
    @     0x7f06f38e0397 openvslam::data::frame::frame()
    @     0x7f06f38af219 openvslam::tracking_module::track_monocular_image()
    @     0x7f06f38a1e3c openvslam::system::feed_monocular_frame()
    @     0x555c348fc455 _ZNSt6thread11_State_implINS_8_InvokerISt5tupleIJZ13mono_trackingRKSt10shared_ptrIN9openvslam6configEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESG_SG_jbbbSG_EUlvE_EEEEE6_M_runEv
    @     0x7f06f0a266df (unknown)
    @     0x7f06f35df6db start_thread
    @     0x7f06f0481a3f clone
Aborted (core dumped)
```

which I believe arise from [this line](https://github.com/surfii3z/openvslam/blob/f2192b0b1f487d234f5ca9ca4142813438615b47/src/openvslam/feature/orb_extractor.cc#L159).

If anyone could help with the debug, I would really appreciate.

surfii3z, 2020-09-15:
One thing that I know is if I comment [line157 ~ 167](https://github.com/surfii3z/openvslam/blob/f2192b0b1f487d234f5ca9ca4142813438615b47/src/openvslam/feature/orb_extractor.cc#L157).

The function gpuOrb.join() at [line159](https://github.com/surfii3z/openvslam/blob/gpu_dev2/src/openvslam/feature/orb_extractor.cc#L159) will ALWAYS be executed two times before the program terminate with the same error. 
`gpuOrb.join(descriptors_at_level);`

If I change [line 153](https://github.com/surfii3z/openvslam/blob/f2192b0b1f487d234f5ca9ca4142813438615b47/src/openvslam/feature/orb_extractor.cc#L153) from 
`gpuOrb.launch_async(blurred_image, keypts_at_level.data(), keypts_at_level.size());`
to
`gpuOrb.launch_async(blurred_image, keypts.data(), keypts.size());`
The above error disappeared. However, it could not track the feature (which is expected).

My opinion is that it is probably something about keypts and keypts_at_level that behave differently from ORB_SLAM2 code (I used the exact same code (?)).