xdspacelab/openvslam #264: Error during viewer::create_menu_panel() right after start

tiziano-v, 2020-03-05:
Hi openvslam Team!

I'm trying to run KITTI Odometry dataset example in visual studio 2019, but right after I run the code the pangolin viewer crash at create_menu_panel().  In particular it generate an access violation exception when it tries to initialise a new var (the menu_follow_camera_). I only started yesterday to work with this project, so i'm not very into it, but in debug mode the exception is thrown here:
 in glfont.h   inline float Height() const {
        return font_height_px;
    }
Unhandled exception thrown: read access violation.
**this** was 0xFFFFFFFFFFFFFFF7.

Today i'll try to figure out more details about, but if you have already encountered this problem and you can help me i'll really appreciate that!
Thank you!

tiziano-v, 2020-03-05:
However i installed openvslam by means https://github.com/reunanen/openvslam-build-vs
and I'm using an NVIDIA graphics cards.

tiziano-v, 2020-03-05:
it seems that context->font.reset(new GlFont(AnonymousPro_ttf, 15)); is not called. how do i specify the font?

reunanen, 2020-03-06:
@tiziano-v I can't reproduce this problem. Trying with VS 2019 and the KITTI Odometry example. Can you can think of anything else that might be relevant in your setup?

tiziano-v, 2020-03-08:
I found out that there was a conflict with vcpkg. I solved the problem removing pangolin from vcpkg and rebuild the solution. sorry, my bad! ; )

closed 2020-03-08 by tiziano-v