xdspacelab/openvslam #266: Camera configuration in .yaml

IaroslavS, 2020-03-05:
Hi.
I've got some dataset of RGB images from stereo camera.
I have the following parameters of the stereo camera

`left:
  camera_model: pinhole
  distortion_model: radtan
  intrinsics: [913.2042513978413, 913.2042513978413, 1007.7655841190004, 564.3105692836759]
  distortion_coeffs: [0.022560997024600776, -0.02002663234521112, 0.0, 0.0, 0.0]
  resolution: [2048, 1280]
  T:
  - [0.9999172576977227, -0.0009141757065997424, -0.01283129147995608, 0.0]
  - [0.0009141004475619156, 0.9999995821413015, -1.1730059856783346e-05, 0.0]
  - [0.012831296841625078, -3.015437292225309e-19, 0.9999176755220213, 0.0]
  - [0.0, 0.0, 0.0, 1.0]
left_rect:
  camera_model: pinhole
  intrinsics: [363.328113923394, 363.328113923394, 389.97408721460715, 218.60497877600005]
  resolution: [800, 500]
  P:
  - [363.328113923394, 0.0, 389.97408721460715, 0.0]
  - [0.0, 363.328113923394, 218.60497877600005, 0.0]
  - [0.0, 0.0, 1.0, 0.0]
right:
  camera_model: pinhole
  distortion_model: radtan
  intrinsics: [914.1260124304225, 914.1260124304225, 998.3512996151677, 546.5636886196756]
  distortion_coeffs: [0.02222741631349973, -0.01873536772011133, 0.0, 0.0, 0.0]
  resolution: [2048, 1280]
  T:
  - [0.9999673065543186, -0.0024276952463401055, 0.007713113398117066, 0.0]
  - [0.002377042690262067, 0.9999755956054291, 0.0065694643324958136, 0.0]
  - [-0.00772887382158533, -0.006550915154248347, 0.999948673692851, 0.0]
  - [0.0, 0.0, 0.0, 1.0]
right_rect:
  camera_model: pinhole
  intrinsics: [363.328113923394, 363.328113923394, 389.97408721460715, 218.60497877600005]
  resolution: [800, 500]
  P:
  - [363.328113923394, 0.0, 389.97408721460715, -144.6075556826869]
  - [0.0, 363.328113923394, 218.60497877600005, 0.0]
  - [0.0, 0.0, 1.0, 0.0]`

I've made the following configuration in .yaml file for openVSLAM, but I don't know, what value should I put for "Camera.focal_x_baseline", "StereoRectifier.R_left" and "StereoRectifier.R_right:".
`# intrinsic parameters after stereo-rectification are written

Camera.name: "pinhole"
Camera.setup: "stereo"
Camera.model: "perspective"

Camera.fx: 363.328113923394
Camera.fy: 363.328113923394
Camera.cx: 389.97408721460715
Camera.cy: 218.60497877600005

Camera.k1: 0.0
Camera.k2: 0.0
Camera.p1: 0.0
Camera.p2: 0.0
Camera.k3: 0.0

Camera.fps: 10.0
Camera.cols: 2048
Camera.rows: 1280
Camera.focal_x_baseline: ?????

Camera.color_order: "RGB"

#======================#
# Stereo Rectification #
#======================#

# original intrinsic parameters (K, D) and stereo-recitification parameters (R)
# matrices (K, R) are written in row-major order

StereoRectifier.K_left: [913.2042513978413, 0.0, 913.2042513978413, 0.0, 1007.7655841190004, 564.3105692836759, 0.0, 0.0, 1.0]
StereoRectifier.D_left: [0.022560997024600776, -0.02002663234521112, 0.0, 0.0, 0.0]
StereoRectifier.R_left: [_0.999966347530033, -0.001422739138722922, 0.008079580483432283, 0.001365741834644127, 0.9999741760894847, 0.007055629199258132, -0.008089410156878961, -0.007044357138835809, 0.9999424675829176_]
StereoRectifier.K_right: [914.1260124304225, 0.0, 914.1260124304225, 0.0, 998.3512996151677, 546.5636886196756, 0.0, 0.0, 1]
StereoRectifier.D_right: [0.02222741631349973, -0.01873536772011133, 0.0, 0.0, 0.0]
StereoRectifier.R_right: [_0.9999633526194376, -0.003625811871560086, 0.007755443660172947, 0.003680398547259526, 0.9999684752771629, -0.007035845251224894, -0.007729688520722713, 0.007064130529506649, 0.999945173484644_]`

And can anybody check my configuration (that is given above) entirely ?
Thanks in advance !

Dung-ND, 2020-03-06:
Hi!
I used this repo to do the calib: https://github.com/ethz-asl/kalibr/wiki/multiple-camera-calibration
For baseline, they noted in the config: focal_x_baseline is -P2[0][3]
So, it's the last one in first row of your 2nd Projection matrix.
Hope it helps!

Dung-ND, 2020-03-19:
Sorry I posted the wrong one. The one that works for me is the ros tool above.

dimaxano, 2020-04-21:
OFFTOP
@IaroslavS I noticed you changed your intrinsic after "stereo rectification". I think it is a wrong step, cause intrinsics in most cases are constant values and should be fixed with changing image resolution. 
So that you should change either image resolution in final config file or camera intrinsic itself

IaroslavS, 2020-04-21:
> OFFTOP
> @IaroslavS I noticed you changed your intrinsic after "stereo rectification". I think it is a wrong step, cause intrinsics in most cases are constant values and should be fixed with changing [...]

Yes, you're right. At the time I wrote this issue I was new to rectification.
Now I clearly see, what should be in config file .yaml, except fields for rectification. That's why I rectify myself images and then put rectified parameters into config file .yaml.
I'm closing this issue, I have no questions on this topic.

closed 2020-04-21 by IaroslavS