xdspacelab/openvslam #268: "error: XDG_RUNTIME_DIR not set in the environment" when running on RGB KITTI Dataset

IaroslavS, 2020-03-07:
Hi.
I've got problem with running openVSLAM on color dataset The KITTI Benchmark.
I ran with with command in terminal:
`./run_kitti_slam     -v /vocab/orb_vocab/orb_vocab.dbow2     -d /dataset/The\ KITTI\ Vision\ Benchmark\ Suite/Odometry/dataset_color/sequences/00     --eval-log     --auto-term     -c /openvslam/example/kitti/KITTI_stereo_00-02.yaml 
`
After that PangolinaViewer is opened, freezes at the start and Frame Viewer outputs message "WAITING FOR IMAGES" terminal showed errors:

[2020-03-07 11:35:43.106] [I] loading ORB vocabulary: /vocab/orb_vocab/orb_vocab.dbow2
[2020-03-07 11:35:43.502] [I] startup SLAM system
[2020-03-07 11:35:43.502] [I] start mapping module
[2020-03-07 11:35:43.502] [I] start global optimization module
error: XDG_RUNTIME_DIR not set in the environment.
(PangolinViewer: Frame Viewer:379): dbind-WARNING **: 11:35:43.585: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-cegF9TaD4b: Connection refused

At the same time I can successfully run openVSLAM on gray dataset with the same messages in terminal:

error: XDG_RUNTIME_DIR not set in the environment.
(PangolinViewer: Frame Viewer:387): dbind-WARNING **: 11:41:41.386: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-cegF9TaD4b: Connection refused

adizhol, 2020-07-19:
@IaroslavS How did you solve this?
Thanks

IaroslavS, 2020-07-20:
> @IaroslavS How did you solve this?
> Thanks

I solved problem with running on RGB dataset not on KITTI Dataset, but on my own dataset.
You should change `Camera.color_order` in config file .yaml from `Gray` to `BGR` or `RGB`.
If you still have problems write here.

closed 2020-10-13 by IaroslavS