xdspacelab/openvslam #270: Cmake error

95ImadEL, 2020-03-09:
Hello,

Can you help me to solve this error. 

`CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
DBoW2_LIBRARY
    linked by target "openvslam" in directory /home/Me/Desktop/openvslam/src/openvslam

-- Configuring incomplete, errors occurred!
See also "/home/Me/Desktop/openvslam/build/CMakeFiles/CMakeOutput.log".
See also "/home/Me/Desktop/openvslam/build/CMakeFiles/CMakeError.log".`

Thank you in advance :)

Shashika007, 2020-03-10:
@95ImadEL libyaml-cpp library may be conflict between two versions. 
if you are running the ros version, try to link your yaml-cpp library from absolute path inside openvslam/ros/openvslam/src/CMakeList.txt. 

example : 

target_link_libraries(run_custom_slam PRIVATE        
${Pangolin_Viewer_LIB} ${Pangolin_LIBRARIES} "/home/your-username/yaml-cpp/build/libyaml-cpp.so" ${Boost_LIBRARIES})

If it does not help, please share how you build the openvslam. your cmake configuration.

95ImadEL, 2020-03-10:
Thank you @Shashika007 it solves the problem

closed 2020-03-10 by 95ImadEL