xdspacelab/openvslam #272: download and installing eigen from source

vishnujachand, 2020-03-11:
cd /path/to/working/dir
gives me No such file or directory.. What to do?

shinsumicco, 2020-03-12:
Please replace `/path/to/working/dir` with your arbitrary directory (e.g. `${HOME}/Documents/`).

vishnujachand, 2020-03-12:
Madam,
Can i do it with cd /home/username
I am new to ubuntu terminal..please help

elfelround, 2020-08-12:
since i did git pull in root i will be doing cd to that directory that was cloned, i do fint this to be a feasible question, conssidering working directory for ros is the src directory

elfelround, 2020-08-12:
as you can see these are installing x from sourcecode
[in the opencv install from sourcecode you can have a better idea of the use of that directory](https://linuxize.com/post/how-to-install-opencv-on-ubuntu-18-04/)