xdspacelab/openvslam #273: Python Bindings

Zumbalamambo, 2020-03-12:
Is there any Python bindings for Openvslam?

stolpa4, 2020-03-26:
++ Is there someone writing the bindings? I can contribute

JackCai1206, 2020-05-27:
I currently have fork here https://github.com/JackCai1206/openvslam that implements some pybind11 bindings around system.h.
It's really messy right now, since I this is only made for a another project I'm working on. But I'm also interested in making this more readable and universal purposed.

Zumbalamambo, 2020-05-29:
@JackCai1206 Wow!... Awesome... Does the localizer work with the python binding?

JackCai1206, 2020-05-29:
here are all the bindings, its very incomplete and is just a start https://github.com/JackCai1206/openvslam/blob/master/python/bindings.cc, right now I can only feed frames into the slam object and get camera pose. 
Looking at `run_camera_localization.cc` in the examples, it seems like on default, the SLAM object does mapping and localization at the same time, if you have a map database file and only want localization, you need to do `SLAM.load_map_database(map_db_path);` and then `SLAM.disable_mapping_module();`

kevkid, 2020-10-16:
@JackCai1206 

> here are all the bindings, its very incomplete and is just a start https://github.com/JackCai1206/openvslam/blob/master/python/bindings.cc, right now I can only feed frames into the slam object and [...]

do you have any examples? do I just build the docker in the same way as the official repo does?

kevkid, 2020-11-19:
@JackCai1206 
I am running into an issue when importing the library:
`ImportError: libopenvslam.so: cannot open shared object file: No such file or directory`
even though under `build/lib` there are the following files:
`libopenvslam.so  libsocket_publisher.so`
to generate these files I ran `sudo ./get-python-bindings.sh`
Any idea why I would be getting this error? I have also appended the full path directory to build/lib