xdspacelab/openvslam #283: Localization mode‘s track time significantly longer than the SLAM  mode's track time

zlgybz, 2020-03-28:
Why is the Localization mode‘s track time significantly longer than the SLAM  mode's track time?
And when using Localization mode,than enable mapping function，Previous maps and keyframes will chang abnormally

otri, 2020-03-30:
I'm not sure of the specific implementation here, but localization is generally a much harder problem of discovering your point of reference.   Once you have your 3D position in space, then differential updates are quick and easier to compute.  Therefore SLAM mode tracking will be far quicker than localization mode.

stolpa4, 2020-03-30:
@otri Can you share some information? I thought that localization is just a part of SLAM (SL without AM). And the localization mode does not imply any map management. So it generally should be faster. What do I miss?

mirellameelo, 2020-05-15:
@stolpa4 Hi! Usually, there are two modes: SLAM and localization. SLAM is responsible for building the map, and this is possible by mapping and localizing simultaneously. Once you ALREADY have that map, it is expected the system to be able to load it and localize the camera pose into that (localization mode).

Zumbalamambo, 2020-05-16:
@mirellameelo what does the `--mapping` flag in the `run_video_localization` do?

mirellameelo, 2020-05-16:
@Zumbalamambo Once you set the --mapping flag, the map you used as input can be extended/updated/modified. 
It may increase time, but I don't know how much, sorry.

mirellameelo, 2020-05-16:
@Zumbalamambo Once you set the --mapping flag, the map you used as input can be extended/updated/modified. 
It may increase time and precision, but I don't know precisely this answer... =/

rebellojason7, 2020-06-19:
I guess the main problem the question is addressing is that localization should be a faster process, assuming you have already localized in the previous frame. But when running purely the localization mode (without updating the map. i.e. disabling any refinement to the already built map), it still takes longer than when running in SLAM mode, which does not make sense to me