xdspacelab/openvslam #285: *** Error in `./run_video_slam': double free or corruption (out): 0x00007f285c80ae40 ***

heshanchuan, 2020-03-30:
When I run the code after making the code successfully. There is problem!
run :
`./run_video_slam -v ./orb_vocab/orb_vocab.dbow2 -m ./aist_living_lab_1/video.mp4 -c ./aist_living_lab_1/config.yaml --frame-skip 3 --no-sleep --map-db map.msg
`

```
*** Error in `./run_video_slam': double free or corruption (out): 0x00007f285c80ae40 ***
...
Aborted (core dumped)

```

heshanchuan, 2020-03-30:
I try the solution of issue [21](https://github.com/xdspacelab/openvslam/issues/21)
it works!
as follows:
`
I've found a solution，configure and rebuild g2o and OpenVSLAM with -DBUILD_WITH_MARCH_NATIVE=OFF option for cmake.`

songuke, 2020-04-13:
For me it was due to g2o is compiled with -DBUILD_WITH_MARCH_NATIVE=OFF but openvslam is compiled with -DBUILD_WITH_MARCH_NATIVE=ON. 
I rebuild g2o with

cmake \
    -DCMAKE_BUILD_TYPE=Release \
    -DCMAKE_INSTALL_PREFIX=/usr/local \
    -DCMAKE_CXX_FLAGS=-std=c++11 \
    -DBUILD_SHARED_LIBS=ON \
    -DBUILD_UNITTESTS=OFF \
    -DBUILD_WITH_MARCH_NATIVE=ON \
    -DG2O_USE_CHOLMOD=OFF \
    -DG2O_USE_CSPARSE=ON \
    -DG2O_USE_OPENGL=OFF \
    -DG2O_USE_OPENMP=ON \
    ..

and rebuild openvslam then it works. It seems the flag value BUILD_WITH_MARCH_NATIVE of openvslam and g2o needs to match.

closed 2020-04-18 by shinsumicco