xdspacelab/openvslam #286: Running headless?

jagauthier, 2020-03-31:
Greetings,

 This is more of a support issue than a bug.  It may be completely a misunderstanding on my part on how this goes together as well.   What I'd like to be able to do is grab video from /dev/videoX and then get an estimated position from that (x,y, and maybe z) and then pass that information to a different program/controller.  Can anyone give me some pointers on how that would be possible?

Thanks!

Shashika007, 2020-04-06:
@jagauthier You can easily run the openvslam (ROS). To get the estimated position of frame, you can use, 
 auto camera_pose = map_publisher_->get_current_cam_pose().inverse().eval();
You can extract pose and the rotation out from that matrix. 
Then you can use that pose to publish to some controller/program which can subscribe that topic.

jagauthier, 2020-04-06:
@Shashika007 Thank you!  Can I do it without ROS?  Probably not. :)

Shashika007, 2020-04-07:
@jagauthier Yes you can. But, it is easier with ROS

jagauthier, 2020-05-25:
Greetings!  This topic is a little old.  But due to other things I have not been able to focus on this.
The line of code you shared:
`auto camera_pose = map_publisher_->get_current_cam_pose().inverse().eval();`

Where is this happening?  
I've launched SLAM (as a test)
rosrun openvslam run_slam -v orb_vocab.dbow2  -c TUM_RGBD_mono_1.yaml

I have a video, but I don't have a map being generated (yet).  Also, I don't see any new topics being published, so I can't seem to access that map using a topic either.

mirellameelo, 2020-05-27:
@jagauthier check out #225 ! 
There is a way to publish the pose data. 
ps: if you don't want to use the entire branch  "[ros2_odometry_example](https://github.com/klintan/openvslam/tree/ros2_odometry_example)", but simply add the modifications in the run_slam.cc or run_localization.cc code, remember that it is also necessary to include the ROS packages used in the CMakeLists file (ros2/src/openvslam) before building the Openvslam packages.

jagauthier, 2020-05-27:
Thanks! I will definitely check that out. I am trying to implement [PR #276](https://github.com/xdspacelab/openvslam/pull/276/) based on our discussion [#308](https://github.com/xdspacelab/openvslam/issues/308).